carlosperate/ardublockly #13: start.py error in windows 7 32bits

RubensQRZ, 2015-06-08:
I tried to work with Ardublocky in a  machine with windows 7-32bits and I  got this errors when trying do run start.py:

Python 2.7.8 (default, Jun 30 2014, 16:03:49) 
[MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.

>>> ================================ RESTART ================================
>>> 
Running Python version 2.7.8

======= Loading Settings =======

Settings loaded from:
	C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.ini

The provided compiler path in the settings file is not valid:
	None
Default compiler path set:
	None

Traceback (most recent call last):
  File "C:\Users\Rubens\dbk_over_ardublockly\start.py", line 120, in <module>
    main(sys.argv[1:])
  File "C:\Users\Rubens\dbk_over_ardublockly\start.py", line 89, in main
    ArdublocklyServer.ServerCompilerSettings.ServerCompilerSettings()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.py", line 61, in __new__
    cls.__singleton_instance.__initialise()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.py", line 77, in __initialise
    self.read_settings()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.py", line 509, in read_settings
    self.set_serial_port_from_file(settings_dict['arduino_serial_port'])
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.py", line 350, in set_serial_port_from_file
    self.populate_serial_port_list()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\ServerCompilerSettings.py", line 401, in populate_serial_port_list
    port_list = ArdublocklyServer.SerialPort.get_port_list()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\SerialPort.py", line 20, in get_port_list
    iterator = get_port_complete_list()
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\SerialPort.py", line 13, in get_port_complete_list
    return sorted(port_list)
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\PySerialListPorts\list_ports_windows.py", line 232, in comports
    yield string(port_name_buffer), string(szFriendlyName), szHardwareID_str
  File "C:\Users\Rubens\dbk_over_ardublockly\ArdublocklyServer\PySerialListPorts\list_ports_windows.py", line 46, in string
    return ''.join(s)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe7 in position 0: ordinal not in range(128)
>>>

carlosperate, 2015-06-08:
Hi Rubens,

I'm unable to replicate this issue in windows 7 32 bit running on a virtual machine. Could you give me a bit more info about your python instillation?
Where does the `RESTART` message come from? are you running the python interpreter inside some IDE (like visual studio)? What COM ports are available in your computer? (you can use the Arduino IDE to check that)

RubensQRZ, 2015-06-09:
Hi Carlos, 

I will try to answer some of these questions tomorow, because i don't have acces to that machine now. I'm working with ArduBlockly  in two machines with windows 7 64bits and I had no problem like that. I used the same procedure  in all  machines: I Installed python 2.7.8, after that I downloaded  ArduBlockly and  closure-library from github, unziped this into my users directory (I have admin rights in all these machines)  and run start.py.

Thank you.

Rubens.

RubensQRZ, 2015-06-10:
Hi Carlos, 

The only COM port avaiable in the machine is COM1 (that's what  the arduino IDE shows me) 

I will describe what did i do to  workaround (i dont know if that is the write word to "escape the problem for now")

I changed the  "string buffer"  in  "list_ports_windows.py" to this:
_________________
def string(buffer):
    s = []
    return ''.join(s)
_________________

I Know that this doesn't solve the problem, but  I can go on with my work  for now in this machine.

Thanks.

RubensQRZ, 2015-06-11:
Hi Carlos.

I "asked" the program to print each "c in buffer", using this line command:

		print (chr(c & 0xff))

And that's what o got:

Settings will be set to the default values.
L
P
T
1
A
C
P
I
\
P
N
P
0
4
0
1
\
4
&
E
5
A
C
E
E
3
&
0
L
P
T
1
P
o
r
t
a
 
d
e
 
I
m
p
r
e
s
s
o
r
a
 
E
C
P
 
(
L
P
T
1
)
C
O
M
1
A
C
P
I
\
P
N
P
0
5
0
1
\
1
C
O
M
1
P
o
r
t
a
 
d
e
 
c
o
m
u
n
i
c
a
ç
ã
o
 
(
C
O
M
1
)

carlosperate, 2015-06-16:
I was finally able to get around to look into this, thank you for all the information provided. Could you check if the latest commit to master fixes this issue for you?

carlosperate, 2015-06-17:
There has been some changes in the python code that might (hopefully not) affect your testing.
I'd recommend to check commit 452fafd161655ba91adae155a1d0a11f62abec29 first in case master gives you an unrelated problem. 

You can download the code directly from this [link](https://github.com/carlosperate/ardublockly/archive/452fafd161655ba91adae155a1d0a11f62abec29.zip) (however, the closure library submodule will be empty and needs to be added if you want to test it properly), or if you are using git you can do:
```
git fetch origin
git checkout 452fafd161655ba91adae155a1d0a11f62abec29 
```

carlosperate, 2015-06-29:
This should be fixed from commit https://github.com/carlosperate/ardublockly/commit/caeac6226e4cd5fffa3c8cb5deb3ddf983fa1a32, however I will wait until more testing can be done before I close the issue.

carlosperate, 2016-03-27:
Since we've moved to use Python 3 for the executables this should no longer be an issue. Please do let me know if you still encounter this using Python 2.

closed 2016-03-27 by carlosperate