carlosperate/ardublockly #15: Preferences window can be triggered multiple times

carlosperate, 2015-07-07:
Not quite a bug in the Materialize framework, but definitely a bug within Ardublockly.

The current Materialize modal functionality does not stop the same modal to be launched multiple times programmatically, so when triggered from the electron menus it creates mutiple dark overlays on top of the content. Then, when the preferences modal is dismissed only the latest overlay is removed.

Can be fixed by either locally tracking the state of the preferences window (kind of hacky) or updating the Materialize framework (already on the works on issue #1647).

carlosperate, 2015-07-13:
This issue is also affecting the online version as the warning message pile up and there is a left overlay blocking all content.

MadTooler, 2016-01-25:
Related...  

The preferences window seems to popup behind the main screen most of the time with Win 7 64b pro and home.  Since you may not realize it is open, a repeat click to open again results in the sequence noted with this issue.

carlosperate, 2016-01-25:
Do you mean the file and directory browser window launched from the
settings?

MadTooler, 2016-01-25:
Yes.  Sorry that was so unclear, but you deciphered my mess correctly.

carlosperate, 2016-01-25:
Thanks for reminding me about that one, I've added it as a separated issue: https://github.com/carlosperate/ardublockly/issues/37

carlosperate, 2016-02-02:
Closed by ea5e146e5e907f9200d9fda263271a1cca9d470f.

closed 2016-02-02 by carlosperate