carlosperate/ardublockly #17: Traducción de Ardublockly al Español

JMRUIZ1, 2015-09-28:
Hola Carlos

He visto tu trabajo sobre Ardublockly y me gustaría realizar la traducción al español de la herramienta. ¿Será posible? ¿Que ficheros debo traducir?. Me gustaría poder utilizarla con mis alumnos y en los cursos de formación que imparto sobre Entornos de Programación de Arduino

También estoy interesado en una versión directamente ejecutable desde Windows de la aplicación Ardublockly, ¿existe?.

saludos

Profesor
José Manuel Ruiz Gutierrez
España

carlosperate, 2015-09-28:
In case anybody looking at this issue is wondering, the original question is asking about Ardublockly translations, if its possible and how to do it, in particular a spanish translation.

At the moment the text in the Arduino specific blocks are not included in the language file for Blockly, but all the original blockly blocks have translations.

Basically what first needs to happen for it to be fully translated is to move all literal strings from the files in the [blockly/blocks/arduino](https://github.com/carlosperate/ardublockly/tree/master/blockly/blocks/arduino) folder into the [blockly/msg/messages.js](https://github.com/carlosperate/ardublockly/blob/master/blockly/msg/messages.js) file.

So, for example [this io_digitalwrite block](https://github.com/carlosperate/ardublockly/blob/master/blockly/blocks/arduino/io.js#L25-L37) needs to be changed from:

```js
    this.setHelpUrl('http://arduino.cc/en/Reference/DigitalWrite');
    this.setColour(Blockly.Blocks.Arduino.io.HUE);
    this.appendDummyInput('')
        .appendField('set digital pin#')
        .appendField(new Blockly.FieldDropdown(
            Blockly.Arduino.Boards.selected.digitalPins), 'PIN');
    this.appendValueInput('STATE', Blockly.StaticTyping.BlocklyType.BOOLEAN)
        .appendField('to')
        .setCheck(Blockly.StaticTyping.BlocklyType.BOOLEAN);
    this.setInputsInline(true);
    this.setPreviousStatement(true, null);
    this.setNextStatement(true, null);
    this.setTooltip('Write digital value to a specific Port.');
```

to: 

```js
    this.setHelpUrl('http://arduino.cc/en/Reference/DigitalWrite');
    this.setColour(Blockly.Blocks.Arduino.io.HUE);
    this.appendDummyInput('')
        .appendField(Blockly.Msg.IO_DIGITALWRITE_TITLE)
        .appendField(new Blockly.FieldDropdown(
            Blockly.Arduino.Boards.selected.digitalPins), 'PIN');
    this.appendValueInput('STATE', Blockly.StaticTyping.BlocklyType.BOOLEAN)
        .appendField(Blockly.Msg.IO_DIGITALWRITE_TITLE_2)
        .setCheck(Blockly.StaticTyping.BlocklyType.BOOLEAN);
    this.setInputsInline(true);
    this.setPreviousStatement(true, null);
    this.setNextStatement(true, null);
    this.setTooltip(Blockly.Msg.IO_DIGITALWRITE_TITLE_TOOLTIP);
```

(in this case, there are two strings inside the block, IO_DIGITALWRITE_TITLE and IO_DIGITALWRITE_TITLE_2, this is not a good practice for translations, but once we've got the majority of the strings moved I can look into changing it to a better way)

So, the strings are then moved into the [blockly/msg/messages.js](https://github.com/carlosperate/ardublockly/blob/master/blockly/msg/messages.js) file:

```js
// Meaningful description
Blockly.Msg.IO_DIGITALWRITE_TITLE_TITLE = 'set digital pin#';
Blockly.Msg.IO_DIGITALWRITE_TITLE_TITLE_2 = 'to';
Blockly.Msg.IO_DIGITALWRITE_TITLE_TOOLTIP = 'Write digital value to a specific Port';
```

Once that is done, the [language files](https://github.com/carlosperate/ardublockly/tree/master/blockly/msg/js) can be automatically generated to include all the new strings, and then it's a matter to edit the translation in the respective file, [like this file for the spanish language](https://github.com/carlosperate/ardublockly/tree/master/blockly/msg/js/es.js).

----

Hola Jose, 

Por el momento the texto en los bloques especificos de Arduino no estan incluidos en la traduccion general de Blockly, pero todos los otros bloques tienen traducciones.

Principalmente lo que tiene que pasar primero para poder traducirlo es mover todos los `strings` de texto de los archivos incluidos en la carpeta [blockly/blocks/arduino](https://github.com/carlosperate/ardublockly/tree/master/blockly/blocks/arduino) a dentro del archivo [blockly/msg/messages.js](https://github.com/carlosperate/ardublockly/blob/master/blockly/msg/messages.js).

Por ejemplo, para [este bloque io_digitalwrite](https://github.com/carlosperate/ardublockly/blob/master/blockly/blocks/arduino/io.js#L25-L37) necesitariamos cambiar el siguiente codigo:

```js
    this.setHelpUrl('http://arduino.cc/en/Reference/DigitalWrite');
    this.setColour(Blockly.Blocks.Arduino.io.HUE);
    this.appendDummyInput('')
        .appendField('set digital pin#')
[... 16 lines cut ...]
    this.appendDummyInput('')
        .appendField(Blockly.Msg.IO_DIGITALWRITE_TITLE)
        .appendField(new Blockly.FieldDropdown(
            Blockly.Arduino.Boards.selected.digitalPins), 'PIN');
    this.appendValueInput('STATE', Blockly.StaticTyping.BlocklyType.BOOLEAN)
        .appendField(Blockly.Msg.IO_DIGITALWRITE_TITLE_2)
        .setCheck(Blockly.StaticTyping.BlocklyType.BOOLEAN);
    this.setInputsInline(true);
    this.setPreviousStatement(true, null);
    this.setNextStatement(true, null);
    this.setTooltip(Blockly.Msg.IO_DIGITALWRITE_TITLE_TOOLTIP);
```

(en este caso especifico, hay dos `strings` dentro del bloque, IO_DIGITALWRITE_TITLE y IO_DIGITALWRITE_TITLE_2, lo cual no es una buena practica para las traducciones, pero tan pronto como el texto se haya movido al archivo adecuado me puedo encargar de mejoralo al formato correcto)

Finalmente, esas `strings` se tienen que mover al archivo [blockly/msg/messages.js](https://github.com/carlosperate/ardublockly/blob/master/blockly/msg/messages.js):

```js
// Meaningful description
Blockly.Msg.IO_DIGITALWRITE_TITLE_TITLE = 'set digital pin#';
Blockly.Msg.IO_DIGITALWRITE_TITLE_TITLE_2 = 'to';
Blockly.Msg.IO_DIGITALWRITE_TITLE_TOOLTIP = 'Write digital value to a specific Port';
```

Una vez esto este hecho, los [archivos de lenguaje](https://github.com/carlosperate/ardublockly/tree/master/blockly/msg/js) pueden ser automaticamente generados para incluir las nuevas `strings`.  Despues simplemente es cuestion de editar la traduccion en los respectivos archivos, en este caso seria [este archivo para la lengua española](https://github.com/carlosperate/ardublockly/tree/master/blockly/msg/js/es.js).

SebCanet, 2015-09-28:
Hell everybody,
thanks a lot for your work on Blockly. I tried to create a wiki about this particular point :
http://tic.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks

Tell me if it's undestandable. There's a lot of file ot change on BlocklyDuino about this point.

Thanks.

carlosperate, 2015-12-21:
The first step for internationalisation has been implemented in commit https://github.com/carlosperate/ardublockly/commit/c66958626cf8657aa9491d1961e64e876574d1ae.

So far this will only trigger a language change in the core blockly blocks, this is what is left doing (I'll be using this for work tracking):
- [ ] Move the arduino specific blocks strings to a translation file
- [ ] Move all the text displayed in the main web interface to a translation file
- [ ] Extended the server settings to store the language selection
- [ ] Update the front en code to retrieve the store language setting from the server
- [ ] Translate all strings to the Spanish language

isokar, 2016-09-30:
Hi,
I was wondering where were the question of internationalisation? I propose my help for a french version of it.
and thanks again for the great tool

carlosperate, 2016-10-01:
Hi @isokar, I am not quite sure I understand your question, could you rephrase it? Are you wondering how to change the language in the Ardublockly application?
We have a base for the french translation (as you can see in [blockly/msg/js](https://github.com/carlosperate/ardublockly/tree/master/blockly/msg/js) and [blockly/msg/json/fr.json](https://github.com/carlosperate/ardublockly/blob/master/blockly/msg/json/fr.json)), but if there is anything missing or that might require some tweaking contributions are more than welcomed :)