carlosperate/ardublockly #18: Scoop

steve24rdg, 2015-11-07:
I was wondering if in the future there might be a scoop block to run the scoop library so I can run several loops simultaneously?

carlosperate, 2015-11-10:
I am not very familiar with this library, is this the library you are referring to? https://code.google.com/p/arduino-scoop-cooperative-scheduler-arm-avr/

steve24rdg, 2015-11-12:
yes thats the one.
it would be really good if this could be included as a block

SebCanet, 2015-11-12:
Hi,
what do you think about this other multi-tasking library : https://github.com/pietern/avr-tasks ?
I was thinking add it but I have a lot other job to do before...

steve24rdg, 2015-11-13:
Hi
It looks as if it does the same thing. I would not know if one is better than the other though.