carlosperate/ardublockly #22: Easier Custom Block Creation

Immortalin, 2015-12-15:
Currently, creating a custom block requires jumping through a lot of hoops. I noticed that you have a block creator in your TODO.md. Do you think you can elaborate on that? I am building something similar. The end goal is to combine the block factory with [this](https://blockly-demo.appspot.com/static/demos/code/index.html), a browser-based code editor, e.g. ace editor, a visual [tree editor](http://jaredforsyth.com/treed/) for the toolbox.xml for easy addition and removal of blocks and finally automated addition and removal of blocks directly from the browser to remove the need for manual copying and pasting. The problem I have is how do you store and parse javascript code. While it is easy to insert code to `blocks.js` and `python.js` or `arduino.js` in your case by simply appending to the file, it is much harder to remove or update it. I have come up with two solutions to this, 1) Use a tool such as [Esprima](http://esprima.org/) to parse the js functions and insert/remove/update the files similar to a key-value database or 2) Split the blocks into separate files and have build.py concatenate them together at runtime. I do not have much experience with writing databases/kv stores  and I believe the second solution to be easier as you are directly reading/writing to the filesystem.

carlosperate, 2015-12-15:
The idea was to first create it as part of the Blocklyduino [BlocklyduinoFactory](https://github.com/BlocklyDuino/BlocklyDuinoFactory) and then implement a version in Ardublockly that connects to the python server to locally create and modify the files required. There is also a feature in the list (if there isn't, there should have been) to automatically add blocks to the toolbox from a directory (format still needs to be designed, I've just got a general idea in my head), which would be used with this feature to have your blocks appear in the Ardublockly toolbox automatically under a new category.

[BlocklyduinoFactory](https://github.com/BlocklyDuino/BlocklyDuinoFactory) is basically an enhancement of the blocklyfactory to use blockls to also create the generator code. This requires some additional "generator blocks" with more freedom (like being able to type imports, and function calls, etc), and also has to have a connection between the two blockly workspaces to synchronise the new block "features" (inputs, outputs, and all that).

Immortalin, 2015-12-15:
@carlosperate the tool still feels rather clunky. A code editor would be nice. How do you plan to update `blocks.js` and `arduino.js`?

carlosperate, 2015-12-15:
If the user is comfortable enough to write JavaScript code in a editor to generate c code, I don't think they'll have any issues following the instructions to add custom blocks. This tool is mostly done to simplify block creation. If the code required for the new block is complicated enough that it cannot be done using blockly, then once again, the user is probably proficient enough to add custom blocks with the current methods.

> How do you plan to update blocks.js and arduino.js?

Not really planning to edit those files, the new blocks and workspace synchronisation will be implemented in the files located in the root directory of https://github.com/BlocklyDuino/BlocklyDuinoFactory.

Immortalin, 2015-12-15:
But how are you going to implement delete and update of blocks. Manually using an editor?

carlosperate, 2015-12-15:
BlocklyduinoFactory would just offer an option to "download files", which are created the same way than the current "download sketch" and "download blocks xml" features.

Ardublockly has a python server, mean to run locally on the users machine, that will take care of receiving the data to create the local files.

carlosperate, 2017-05-07:
In retrospect my original though of making a blockly factory to generate Ardublockly blocks and generators would probably require more effort than its worth. It's not that difficult, but it's always hard to find free time and there are other user facing features that could have more impact.
So closing this as "won't implement".

closed 2017-05-07 by carlosperate

Immortalin, 2017-05-12:
@carlosperate [this issue](https://github.com/google/blockly/issues/977) might be of interest to you

carlosperate, 2017-05-12:
Thanks @Immortalin, I've been playing with that and similar features for a different project and while I do see a lot of potential I think I need to advance some core features of Ardublockly before start investing the time to implement something similar that could work for the C/C++ language