carlosperate/ardublockly #23: Implement blockly changes into build.py file

carlosperate, 2016-01-15:


syriley, 2016-01-30:
Hi Carlos,

I tried fixing this myself, but got tied up in knots with the build.py. Presumably it's something to do with making sure the static-typing file is included in the build process...

carlosperate, 2016-01-30:
Yeah, been looking into this during the past couple of days, still no luck. Static Typing seems to be included correctly, however I'm getting issues with the blocks file.

carlosperate, 2016-01-30:
It all was caused by small mistakes made by not really knowing how the closure library works when I started developing Ardublockly. Made a few changes to get it working in 154d9f123aa26a92b2bc5d5e052ba6c3cefeafc2, still a few improvements coming to refine a few things.

closed 2016-01-30 by carlosperate