carlosperate/ardublockly #25: Cannot Declare Variable Type at functions

MadTooler, 2016-01-16:
When creating the functions, I did not find a method to declare the types of variables that were to be received.

carlosperate, 2016-01-17:
I need to revisit the function code generator, as there are a few corner cases that might not quite work. But the general idea is that if an argument is set, it will get assigned the type of the value used the first time the variable is set to something.

So in this example, `x` should get automatically assigned to be a `String`, and `y` an `int`:

```
<xml>
  <block type="procedures_defnoreturn" x="97" y="137">
    <mutation>
      <arg name="x"></arg>
      <arg name="y"></arg>
    </mutation>
    <field name="NAME">do something</field>
    <statement name="STACK">
      <block type="variables_set">
        <field name="VAR">x</field>
        <value name="VALUE">
          <block type="text">
            <field name="TEXT"></field>
          </block>
        </value>
        <next>
          <block type="variables_set">
            <field name="VAR">y</field>
            <value name="VALUE">
              <block type="math_number">
                <field name="NUM">0</field>
              </block>
            </value>
          </block>
        </next>
      </block>
    </statement>
  </block>
</xml>
```

MadTooler, 2016-01-17:
Carlos,

That was sort of what I figured you were intending.  As a functional note, if you take your example and simply swap the set values, there seems to be some lingering update to the arduino code.  Both x and y remain as strings until other changes are made, or the rebuild button within the xml window is clicked.

The other issue I could see, this method generates global variables when that may not be most desirable.  It seems to somewhat undercut the need to be passing values to the functions if you already can access them as global.

My comments and suggestions should be taken as they are intended, constructive from a user perspective, without a strong understanding of how the software is developed.  If something I say sounds like I don't know what I am talking about from the programming implementation side, it's because I don't.

I really like what I see here, and hope to be a beneficial resource for your testing and development.  I have been looking for a setup like ardublockly, and am excited to see others with like goals.

Thanks for all your hard work.

carlosperate, 2016-01-17:
> As a functional note, if you take your example and simply swap the set values, there seems to be some lingering update to the arduino code. Both x and y remain as strings until other changes are made, or the rebuild button within the xml window is clicked.

Yes, this is one of the reason I need to update the function generator code, as variables are dealt differently with that generator.

> The other issue I could see, this method generates global variables when that may not be most desirable. 

This is due to a design decision made from Blockly itself, which has been [discussed in the google group](https://groups.google.com/forum/#!searchin/blockly/function$20global/blockly/rGMz3NnFVRI/3gyEFxZAD4wJ) in several occasions. You can see in [this blockly example](https://blockly-demo.appspot.com/static/demos/code/index.html#uq66q6), that it is currently the main way to deal with function arguments, but it is not ideal. At the moment I am very tempted to leave it as it is and maybe look into it again when other higher priority features are fixed/implemented.

> It seems to somewhat undercut the need to be passing values to the functions if you already can access them as global.

This is very true, and doing that might actually make more sense in terms of code behaviour, as a C/C++ function argument scope will ignore the fact that there is a global variable with the same name, so somethings like this:

```c
int x = 0;

int foo(int x) {
	x = 5;
	return x;
}

int main(void) {
	printf("%d ", x);
	foo(x);
	printf("%d ", x);
	printf("%d", foo(x));
	return 0;
}
```

Would output: `0 0 5`, which will be hard to explain to novice programmer.

In fact, this is making me consider to remove the mutator from the functions until a better alternative is formulated.

> hope to be a beneficial resource for your testing and development

You input is greatly appreciated indeed. It's been a while since I was able to test Ardublockly with users in person, and any feedback is more than welcome.

MadTooler, 2016-01-19:
Carlos,

Have you seen this version of blocklyduino:
https://blocklyduinoenhanced.appspot.com/static/demos/code/index.html

It allows for local and global declarations with the variety of types at both as well.  This was more what I was looking to achieve, along with the nice wrapper for the IDE feedback and side by side arduino code that you have created.

Where a more simplified function usage may be more desirable for newbies, the basic function without passing arguments could be used with globals.

carlosperate, 2016-01-19:
Thanks for the link! Very interesting indeed. 

The only issue that I see is that it removes the global scope of the variables, which is relatively easy to achieve, but it still maintains the variables in the name database used in the "variable blocks". So one could create a function with argument `x` and `y`, and then use them outside of the function, which generates the following invalid code:

```C
void do_something(long x, float y) {
}

void setup()
{

}

void loop()
{
  x = "";
  y = "";

}
```
This example seems to break the XML generator as well, so I wasn't able to export it here.

MadTooler, 2016-01-19:
Correct.  However, you can also declare those variables global and use them just as C intended (I think).

This would allow for more accurate code, but also open up the room for error for those that are yet to know what they are doing.  That is where the simple functions without incoming arguments would be simpler for them, then move on to the available advanced with experience.  Maybe a check box for use global or local could do it?

MadTooler, 2016-01-19:
String  x;
String  y;
void do_something(int x, int y) {
}

void setup()
{
  x = "nada";

  y = "nothing";

}

void loop()
{

}

MadTooler, 2016-01-19:
[image]

MadTooler, 2016-01-19:
...not that I generally do or encourage using the same name locally and globally.

carlosperate, 2016-01-22:
The commit linked above removes the capability to add arguments to a function.
Meant to be a temporary workaround until a better approach is defined and implemented.

MadTooler, 2016-01-23:
This seems to clean things up for the time being, but it would be ideal to have the full functionality in the future.