carlosperate/ardublockly #26: Change item/variable name or new not popping up

MadTooler, 2016-01-16:
When using the current ready to run windows version on Win 7 64, the popup window normally seen in blockly, or the ability to type over the label of the variables (default as "item"), does not appear to function.

carlosperate, 2016-01-17:
Looks like prompt is not supported on Electron (the framework used to created the desktop application). So it looks like an HTML implementation will have to be created to fix this.

MadTooler, 2016-01-19:
Work around for now is renaming the variable within the "blocks xml" window.

carlosperate, 2016-01-19:
I've got planned a simple workaround to get that working that should not take much time to implement.

carlosperate, 2016-01-22:
Partial fix to this issue, so I'll leave it opened for now. Turned out to be a bit more complex than anticipated, as I was hoping to not have to modify core Blockly.

The functionality is accessible, but to create a new variable with a custom name an additional step is required, first you create a new variable, then rename it. I'll have to further refactor Blockly to combine both steps into 1 again.

https://github.com/carlosperate/ardublockly/commit/544cdb941949185121b82f0f98df0449ddd54f25

carlosperate, 2016-01-22:
Set up milestone for the version after next, as partial workaround is probably adequate for first release:
- [x] Refactor Blockly to allow async update of variable names
- [x] Update desktop aplication to use Ardublockly HTML modal instead of window.prompt (browser version still uses prompt) 
- [ ] Further refactor Blockly to allow async creation of a new variable name
- [ ] Submit blockly changes as upstream PR

MadTooler, 2016-01-23:
This seems to clean things up for the time being, but it would be ideal to have the full functionality in the future.

carlosperate, 2016-01-24:
Thanks for testing, glad the partial solution works well.