carlosperate/ardublockly #27: Compile check not displaying errors

MadTooler, 2016-01-16:
Even when there are clear errors in the sketch, the return message at the bottom banner still reports that the check was successful from the Arduino IDE.

carlosperate, 2016-01-17:
If you are running ardublockly under windows, are you sure it is pointing to the arduino_debug.exe file and not the arduino.exe?

MadTooler, 2016-01-17:
I know I had the "compiler location" under Edit>Preferences set to the arduino.exe.  I am not sure that is where you meant, but I found in re-running from the ardublockly_run.bat file, I cannot change the compiler location and sketch folder for some reason.  I tried modifying the "ServerCompilerSettings.ini" to the arduion_debug.exe, and now it says "Please select a valid Arduino compiler directory" form within settings.  Did I make a booboo?

MadTooler, 2016-01-17:
Oh, for some reason it was resetting the files to default.

I noticed it was popping up the file window under the ardublockly app when I was trying to set the compiler location.

Now, with the arduino uno still plugged in, it keeps the settings.

The uploading and checking seem to work OK.  The open in arduino IDE was really slow the first time and locked up.  After closing with task manager and restarting, it seems to work.  

Too bad arduino IDE takes long to load in general.  I have noticed It adds a lot of lag on older machines, and could give the user the impression your app may be hanging when it is simply waiting on the IDE.

Thanks.

carlosperate, 2016-01-17:
Make sure you are running version 1.5 or higher of the Arduino IDE as well, ideally the latest one available (currently at 1.6.7).

One current issue with selecting the "arduino compiler location" and the "sketch folder" is that the file browser windows sometimes opens in the background and it is not clear that they have been opened. See if that might have been your problem when trying to select a different location using the settings.

As a general note, the `ardublockly_run.bat` basically executes the `ardublockly.exe` file within the 'arduexec' folder

carlosperate, 2016-01-17:
>Too bad arduino IDE takes long to load in general. I have noticed It adds a lot of lag on older machines, and could give the user the impression your app may be hanging when it is simply waiting on the IDE.

Yes, this is definitely a bit of an issue. Technically, the spinning wheel should indicate that the IDE is still compiling/uploading, but as you said, it can take an excessive amount of time with some machines.

The are currently two possible improvements that can be done, to maintain temporary files between compilations, or to move from the IDE CLI to the more efficient Arduino builder CLI (but with this second route I am not sure what options are available to upload the compiled program).

carlosperate, 2016-01-17:
Looks like this issue can be closed then, however because the "not using the arduino_debug.exe" issue might pop a few time I'll leave this open until I implement a check on the server side.

So what I'll probably do, is if a user selects an "arduino.exe" file for the compiler, it will automatically check if an "arduino_debug.exe" exists on the same directly and use that instead.

MadTooler, 2016-01-17:
Ok. It does seem to work proper now.

You may want to change the instructions you made for setup at:
https://github.com/carlosperate/ardublockly/wiki/Arduino-Compiler-Settings

It was from there that I chose to use the arduino.exe file.  Or, you could just leave the setting as what I was expecting when the field is "Compiler location," where the input is merely the folder, not the file.

carlosperate, 2016-01-17:
Thanks for the letting me know about the instructions, I've updated that line.

Unfortunately the wiki is not as complete as I would like it to be. Please feel free to update it if you find any incorrect information or you'd like to help me improve the, rather limited, documentation. Any contribution will be really appreciated!

carlosperate, 2016-01-18:
Oh, and about setting only the IDE directory instead of selecting the file, that's a good idea, although I'll have to look into how it would be done in OS X, as it can be a bit confusing taking in consideration that .app bundles are basically just folders.

MadTooler, 2016-01-19:
When I was running 1.6.5 yesterday it was working, now upgraded to 1.6.7 and it does not appear to receive IDE output again.  I reset the compiler location and have a fresh install of both ardublockly and arduino.  Confirmed I am pointing to the arduino_debug.exe.

My current work around is to either send to the IDE, or revert back to 1.6.5.  Otherwise, ardublockly hangs in a wait without ability to send out again.

carlosperate, 2016-01-19:
I'll try to test this tonight with 1.6.7

carlosperate, 2016-01-19:
I was able to try 1.6.7 with Ardublockly and had no issues under Windows 10 x64, with the zip version of the Arduino IDE. Could you give me a bit more information about the system you are using?

MadTooler, 2016-01-19:
I am running from the windows installer version of 1.6.7 on win 7 64 pro and home.  Both machines have problems.  My pro just hangs "waiting for the IDE output" when checking or sending direct to arduino after it does load the program properly.  My home machine works ok maybe the first time or so, then does the similar thing.  Really has a hard time if there is a problem with the sketch.

carlosperate, 2016-01-19:
Having the issues only after a successful load can be the key here, I'll try again once I get a chance tomorrow night.

MadTooler, 2016-01-23:
Few more results on this issue...

I tried using the non admin arduino zip 1.6.7 and found same results under Win 7 64b pro as with the arduino under windows installation.

Arduino zip 1.6.6 also had same.  Both 1.6.6 and 1.6.7 will open in IDE just fine, check does not return response from IDE, and upload will upload without a final response from IDE as well.

However, 1.6.5 r5 both zip and windows installation work entirely.

MadTooler, 2016-02-10:
This may be further expanding whatever error I am experience with compatibility of the newer arduino IDE.

When using either the standalone or the browser interface of the latest build, arduino 1.6.7 gives the following error when verifying the blink example:

Unexpected exit error code: 255
Loading configuration... 
Initializing packages... 
java.lang.NullPointerException 
at cc.arduino.contributions.packages.ContributionsIndexer.mergeContributions(ContributionsIndexer.java:119) 
at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:91) 
at processing.app.BaseNoGui.initPackages(BaseNoGui.java:597) 
at processing.app.Base.(Base.java:295) 
at processing.app.Base.guardedMain(Base.java:222) 
at processing.app.Base.main(Base.java:136)

carlosperate, 2016-02-10:
That kind of looks like an error with the board manager. A long shot, but do you experience this issue only when the system is not connected to the internet?

MadTooler, 2016-02-11:
That is connected to internet, no board connected.  I need to check what board was selected.  May have been the default.

carlosperate, 2016-02-21:
I've been using 1.6.7 last week under Windows 10 x64 and was not really able to replicate this issue.
In the computer where you are seeing this, have you installed any additional boards and/or libraries using the board or the library manager? Even if you have uninstalled, reinstalled, or used the zip Arduino IDE versions.

MadTooler, 2016-02-21:
I was thinking about the libraries, but not the boards.  I have two systems acting the same, one with extra boards installed (adafruit and sparkfun), and one without.

Both systems have started to refuse opening 1.6.7 altogether (splash screen then nothing).  Even outside of ardublockly.  I have seen others with the issue, but not a solution.  It seems it may be related.

As for libraries, I have numerous, some of which are experimental.

As a controlled test, I ran from the zip ide under a vmware virtual machine win764b, and it ran fine.  It created the arduino sketch folder and ran like it should.

I then went back to my regular system, renamed my arduino sketch/library folder, and tried to run.  It created the sketch folder, but never started 1.6.7.  Same for 1.6.6.  When I reverted to 1.6.5r2 it ran fine again.

I do not know what all is rattling the system, but my guess is it is not ardublockly or my libraries.

For fun, I also tried disabling my AVG anti virus, still not luck.

carlosperate, 2016-02-21:
There is a chance it could be caused by a malformed board or library package: http://forum.arduino.cc/index.php?topic=321711.0

The Arduino IDE creates user data folders on the system which it would still grab even if you upgrade, maybe version 1.6.5 doesn't parse some of the data in these folders. It would be worth having a google search to find out where exactly were all of these data files are saved (probably in you user `AppData\Local\Arduino15` apart from the Arduino folder in `my documents`, but there might be more locations), and delete them to see if the problems are still present.

MadTooler, 2016-04-10:
Carlos,

I cleaned out the arduino IDE, and all the arduino15 from my system and have had better with the newer versions, but I am still having trouble with no response  when there is a compile error.  Have you had any luck reproducing this problem on your end?

carlosperate, 2016-04-10:
I haven't seen this issue, but it might be because I haven't been using the loading sketches feature too much.

Is this the same error you were describing before? Is it happening always or is it interminent? Sounds like the main problem you have right now is to do with displaying compile errors (telling you it's compiled correctly)? Could it be relaeted to https://github.com/carlosperate/ardublockly/issues/59 ?

Would you mind opening a new issue indicating the steps and outputs you experience? (also things like Ardublockly version, if running the desktop app or just the python server, operating system, and all that) This github issue is currently open as a reminder for me to add a check for `arduino_debug.exe`, and if we discuss things here, things will get mixed up, and I accidentaly overlook this.

MadTooler, 2016-04-10:
It seems to be the same issue as before.  And yes, it does seem related to #59.  I have not tried sending the sketch to the board lately, but the problem seems to be persistent when there is a sketch with compile  errors.  When you check a good sketch, it does receive the all clear response and act as expected.

I am running "ardublockly_Windows32_2016-04-09_00.35_master_1ad1e" in win7 64 pro.  Both the desktop and chrome version from the ardublockly.exe or start.exe files.  I have tried running from the python module, but I do not remember if this occurred there as well.

I will get back to this and better spell out the steps in a new issue as soon as I have the chance.

Thanks.

carlosperate, 2017-05-06:
Closing due to inactivity, please feel free to reopen if you'd like to look into this again, but so far seems to be a combination of #59 + a problem with your Arduino IDE installation.

closed 2017-05-06 by carlosperate