carlosperate/ardublockly #30: AnalogWrite to digital pins?

MadTooler, 2016-01-17:
Any short term plans for expanding the analogue write to include digital/pwm pins?

Is there somewhere besides issues I should post enhancements?

carlosperate, 2016-01-17:
I just realised, the `analogWrite` function should only work with PWM pins (https://www.arduino.cc/en/Reference/AnalogWrite), so this is currently a bug. 

About where to add enhancement suggestions everything should go here, as it is the best way for me to track it and manage it within github. I can add the "enhancement" tag to each individual entry to differentiate them from bug reports.

MadTooler, 2016-01-17:
Thanks again.

When the float, variable naming, and pwm are up and working, I may be able to start using your stuff with some local groups along with my robots.  That would be really cool.

commit b26b7884 (2016-01-17): Fix analogWrite to list only PWM pins and warn if input is out of range.

Fixes #30
closed 2016-01-17 by carlosperate