carlosperate/ardublockly #31: AVG Antivirus Support

MadTooler, 2016-01-17:
When I tried to run Ardublockly with AVG antivirus running, AVG blocked some of the files.  I think it was the "start" file that was the trigger.

Right now I am running within a vmware virtual machine without any AVG.

carlosperate, 2016-01-17:
Is it the `start.exe` file from the `arduexec\server` folder? That would be the python launcher for the server side of the application (technically speaking it could execute any arbitrary code, but that'd be no different than any other interpreter).

I am not sure there is much I can do, other than to suggest to whitelist the application, but could you send me the report if you get it again? It'd be interesting to see why it is being reported.

MadTooler, 2016-01-18:
That's the file.  I reinstalled on another system and it confirmed.  I allowed it and set ardublockly to run with admin privileges and now it is happy.

carlosperate, 2016-01-18:
Does it give you any indication as to why it is being detected as a threat?

MadTooler, 2016-01-18:
Not that I can tell.  It just says unknown.  I think is doesn't like the potential of a server setup that may be a backdoor in disguise.

carlosperate, 2016-01-22:
At think at this moment there really isn't much I can do, closing this for the moment, we'll see if we get more reports.

closed 2016-01-22 by carlosperate