carlosperate/ardublockly #35: Comment popup window jumps over tool menu

MadTooler, 2016-01-23:
The text of the comments window pops up, sort of, over the top of the tool/block menu bar after clicking back on "?" to hide and re-display comments. 

Initial entry:
[image]

1st Click on "?" to close: (BTW, seems related that comments do not show in arduino source code until  the block is moved around)
[image]

2nd click back on "?" to view comment: 
[image]

Additional view and hide clicks respond the same as 2nd until the block is moved with the comment popup exposed then it works along with the comment now present in the arduino source code:
[image]

Once the comment is closed, it will again act the same way with the text top left of window until you drag the block, then, it does pop back in place.  However, when the comment is hidden, there is a brief blip of the text in the correct popup location before it fully closes.  Additionally, removal or changes of the comment do not update in arduino source code until the block is drug around.

carlosperate, 2016-01-24:
Thanks for the detailed report! This was an odd one, and seemed to be a bug with chromium. So, I updated it to the latest version and I haven't been able to replicate again, so hopefully that should have fixed it.

MadTooler, 2016-01-30:
It is no longer bouncing around on the screen, or leaving any artifacts, but it still does not update the arduino source code until I drag around the block.

carlosperate, 2016-01-30:
Thanks for confirming. 

The code generation is a different issue with how often the code is generated, I'll open a different issue for that one.

closed 2016-01-30 by carlosperate