carlosperate/ardublockly #38: Menu items broken

MadTooler, 2016-01-26:
Under Win7 64b pro...

With the latest build, 35c72, the Time, Motors, and Comms block menus are either blank or significantly reduced from prior functions.

The functions menu has functions with input variables available.  Not sure if you are just in the middle of adding the functionality back, or if this was also a side effect of the update.

carlosperate, 2016-01-27:
35c726c50efa8d97334ad6c9b41d55c79061361b comes from the better_types branch (full file name is ardublockly_Windows32_2016-01-26_02.01_better_types_35c72.zip), which is an unfinished feature branch, so it is currently broken. I'm still working on it, testing a couple of things before merging into master.

MadTooler, 2016-01-27:
I figured that was the case, just making sure u were aware.

What about longs and doubles?

carlosperate, 2016-01-27:
Wasn't planning to support them as auto-detected type, but could be added to the "cast" block.

carlosperate, 2016-01-27:
Latest master now has the better_types branch merged in.

closed 2016-01-27 by carlosperate