carlosperate/ardublockly #39: Generic Motor Enhancement

MadTooler, 2016-01-27:
Is there a list of block requests I should point this towards?

In addition to the stepper and servo motor blocks, a generic motor block which at minimum can handle a pin for direction (0,1) and a speed pin (0-255) would be cool.  Maybe another expanded version which handled two motors labelled left and right.

carlosperate, 2016-01-27:
Right now the issue tracker is the best place to request new blocks.

About the dc motor controller, a pwm pin can be used to control speed, but how would the other pin control direction without additional circuitry?

MadTooler, 2016-01-27:
The pins would be driving a motor driver.  Such as the Sparkfun Ardumoto.
The driver has an h-bridge that determines forward/reverse rotation based
upon 0/1 from direction pin.