carlosperate/ardublockly #42: labels editing is skewed

MadTooler, 2016-01-30:
The editing of labels and some variables has the text in an awkward location:
[image]
[image]

carlosperate, 2016-01-30:
Wow, I haven't seen that in a long time. Is that running the desktop application? or is it running in any other browser? Are there any specific steps required to reproduce it?

MadTooler, 2016-01-30:
That was running from the desktop fresh from starting the program.

This is with the latest build.

carlosperate, 2016-01-30:
Yes, I can definitely reproduce it, I'll have to look what change introduced this issue.

carlosperate, 2016-02-05:
Should be back to normal now.

commit 0e8cea0c (2016-02-05): Fix css fro blocklyHtmlInput being overwritten by Materialize.

Fixes #42.
closed 2016-02-05 by carlosperate