carlosperate/ardublockly #45: Current builds missing menus

MadTooler, 2016-02-01:
Win7 64b pro...

The builds from 01/30/2016 and 01/31/2016 both are missing the blocks menu and give an error of "invalid xml" when an example file opening is attempted.

MadTooler, 2016-02-02:
The ardublockly_Windows32_2016-02-02_00.22_master_ea5e1 build also has this problem.

Several other problems are also present, but I am assuming they are all related to whatever is the root of the menus issue, so I will not bother going into detail at this time.

carlosperate, 2016-02-02:
I see, the fix from https://github.com/carlosperate/ardublockly/commit/bd5bfff66a03c8a61c96dd75c8e2d871535acccc was inadvertently removed automatically by the changes to the build file update this weekend (still waiting for the original fix to be merged into upstream Blockly). It was already patched manually in the build file on commit https://github.com/carlosperate/ardublockly/commit/26b84b9a479e3bf886284178c36a4c8a2baa9a7f, but it has somehow missed to add the updated `blockly_uncompressed.js` file with the fixed embedded inside.

So if you want to get a running version until I recompile the compressed files, you can edit this line [/blockly/blockly_uncompressed.js#L4](https://github.com/carlosperate/ardublockly/blob/master/blockly/blockly_uncompressed.js#L4) with the following:
```
var isNodeJS = !!(typeof module !== 'undefined' && module.exports &&
                  typeof window === 'undefined');
```

MadTooler, 2016-02-02:
That line of code fixed it.
Thanks.

carlosperate, 2016-02-02:
Thank you for catching all this issues early on! All these have definitely helped me a lot.

closed 2016-02-02 by carlosperate

MadTooler, 2016-02-02:
My pleasure.  I am still very excited to see the further development of a really cool tool.  I have been mentioning it to several local groups, and hope to show a version with some of my own blocks later this month with even more.

Thanks again.

MadTooler, 2016-02-02:
BTW, I was checking out codebender.cc.  They have a pluggin for chrome that allows direct (or at least I think) access to the arduinos without use of the IDE.  I didn't know if this was of any use to you.

carlosperate, 2016-02-02:
Unfortunately chrome pluggings cannot be integrated in Electron, so it's not really a plausible option. Apart from that, I believe the compilation is performed on their servers, and one of the main requirements that Ardublockly had when it was first conceived was to be able to work completely offline (a feature I would still like to maintain). The serial connection could be useful, although only the front end could really be ported, and that is generally the simplest component.

Thank you for the tip though, it's been a while since I had a good look at what they were doing, and it's definitely interesting to check out. It's also quite awesome that they open source so much of it!

MadTooler, 2016-09-06:
Carlos,

I am not sure if this is newer than when I mentioned codebender before, but here is a link to their blockly version: http://blockly.codebender.cc/