carlosperate/ardublockly #47: Problem with <  and > logic operators during code generation,

RubensQRZ, 2016-02-04:
Dear, Carlos.

There is a problem with the code generation when we use the > or < operators,

after the first time you choose one of them in the dropdown list,  you get the simbols  &gt;  and  &lt; into the code instead of the simbols < or >.  I'm workin in that fork I told you and I need solve this problem that also happens in ardublockly.

Best Regards

Rubens.

carlosperate, 2016-02-04:
I assume you mean `&gt;`  and  `&lt; `, yes the conversion between text and HTML is definitely messing with those symbols, I'll have a look.

(Btw, you can see that the build.py has been updated :)

RubensQRZ, 2016-02-05:
Thank you so much,  Carlos.

I have not seen the build.py.

I will check

Best Regards

Rubens.

carlosperate, 2016-02-21:
Sorry it took so long to finally look into this. Should be good to go now :)

closed 2016-02-21 by carlosperate

RubensQRZ, 2016-02-23:
Thank you so much, Carlos

Best regards

Rubens

RubensQRZ, 2016-03-01:
Dear, Carlos.

With the soluction to this problem, another problem arised.

We can not use the symbols  "<" and ">" in the string arguments of addinclude function. To workaround,  i'm now using the html entities in the second argument of this function.

Best regards.

Rubens.

carlosperate, 2016-03-02:
I see, so that's why the original HTML sanitation was made! I remembered there was a reason, but couldn't find incorrect code generated in my tests for this fix (I should have been looking for missing code, not just scape characters!). 

Please keep in mind that the Arduino code generated and loaded into the IDE (for opening the code, verifying or uploading) bypasses all of this, and therefore should be correct. So if you add the HTML scape characters to the javascript generator function to include them in the C code, then it will be displayed correctly on the application, but shouldn't work on the Arduino IDE (unless you are copying and pasting).

RubensQRZ, 2016-03-02:
Thank you for de warning, Carlos, I had not yet tried to send the code to the IDE. 

Best regards

commit 8e31049e (2016-02-21): Removes unnecessary HTML scape chars sanitation on Arduino code generation.

Originally placed before the code diff feature was added, it is no longer required. This was also causing issues with code being displayed with the scape characters instead of HTML unsafe symbols.

Fixes #47.