carlosperate/ardublockly #48: Servo Pins, need more

MadTooler, 2016-02-15:
The servo motor module is limited in the pins you can choose.  The servo library allows for almost all of the pins, including the analog, to be used for servos.  They are not limited to PWM pins.

carlosperate, 2016-02-15:
I wrongly assumed it used the PWM peripheral to generate the signal. However, it looks like while it can use any pin, it will disable the PWM capabilities, with a special rule that in the mega it only interferes with PWM if using more than 12 servos...

https://www.arduino.cc/en/Reference/Servo

This logic implementation for the block warnings might require some rework around some areas. Do you need to pin access soon, or could it wait a couple of weeks to implement the change with the correct warnings?

MadTooler, 2016-02-16:
I did not know that about the PWM limitation.  Glad you looked into it.

I am fine for right now.  Probably going to use pins 9 and 10, for the most part.

Thanks again.