carlosperate/ardublockly #49: Add on Blocks usage

MadTooler, 2016-02-18:
How is it we enable the add on blocks like the grove?

I saw some mention of a function to call from the module, I think, but I am unsure where/how that is done.

I have been able to open open the examples of grove and play with the blocks, but I am not sure how to get the menu item in the tool bar to show.

carlosperate, 2016-02-18:
I been working on a few Grove blocks for a couple  of days for a quick demo I wanted to do this week, so they’ve been purposely “hidden” for general users as some of these blocks are likely to change, and then any saved XML using them would stop working after that.

If in the mean time you want to play with them, on your own risk, you have two options:

* Installing node.js you can run the desktop application in “developer mode”, so an extra drop down menu appears in the menu bar with item that lets you open the “load extra blocks” window.
* You can just manually open this window by executing this line of code on the front end: `Ardublockly.openExtraCategoriesSelect() `

I generally try to have meaningful information on my commit messages, so in the future you might find useful to have a look over there to see if there is any extra info: https://github.com/carlosperate/ardublockly/commit/192bb341309b921ee942484eb81a48c5b3aea56b

MadTooler, 2016-02-18:
> I generally try to have meaningful information on my commit messages, so in the future you might find useful to have a look over there to see if there is any extra info: 192bb34

I understood you were intentionally leaving those blocks hidden, and why.  I also had seen your commit notes, which I am sure are very useful to someone that actually knows what they are doing, but I am not much more than a tourist when it comes to software development. 

My real goal here is working on some of my own blocks and testing them.  I was only looking to the grove blocks as examples.  I am not sure how or where to implement either of the solutions you mentioned, so I probably will just wait until you have the user block add in feature ready for full access.

Thanks.

carlosperate, 2016-02-18:
If you still want to give it a go, these would be the steps using the normal desktop application + a browser like Chrome or Firefox:

* Run the Ardublockly application
* Open a web browser on the same computer Ardublockly is running and go to the ` http://localhost:8000/ardublockly/ ` address
* Once the Ardublockly web interface is fully loaded, in the address bar type ` javascript:Ardublockly.openExtraCategoriesSelect() `

This should open a modal window (like the preferences) allowing you to add the Grove category to the toolbox.

MadTooler, 2016-02-19:
Thanks.  That worked for the browser version.  When I get a free minute I can try making my new blocks.

Interesting that on my system, chrome would not copy/paste the "javascript:" before the "Ardublockly.openExtraCategoriesSelect()."  I had to type that in or google simply found this page in a search.

carlosperate, 2016-02-19:
Yes, it's most likely done for security, as you could in essence execute any JavaScript using that, you might be able to trick a user to use it for nefarious reasons.

carlosperate, 2016-02-19:
Also, as a general warning, none of the changes made to the blockly fork for this new feature have not been compiled yet

closed 2016-02-19 by carlosperate