carlosperate/ardublockly #51: Uploading the sketch to the Arduino in UBUNTU 14.04 LTS

RubensQRZ, 2016-03-02:
Dear Carlos, 

When I try to upload the sketch to the Arduino using UBUNTU 14.04 LTS, I get this error:

[image]

Best regards

Rubens

carlosperate, 2016-03-02:
Could you copy the contents of the `ArdublocklySketch.ino` here? It would also be useful if you could upload the `ardublockly.log` (and `ardublockly2.log` if it exists) files somewhere where I could download them.
Which version of the Arduino IDE are you using?

RubensQRZ, 2016-03-09:
Hi Carlos, 

I did not find the ardublockly.log file

The Arduino IDE version is – 1:1.0.5+dfsg2-2

I copied the CLI command: 

/usr/bin/arduino --upload --port /dev/ttyACM0 --board arduino:avr:uno /home/labvad/DB4k/DB4KA915/ArdublocklySketch/ArdublocklySketch.ino

to the Terminal, and I got the same problem: I receive a window with the message: bad file selected: Processing can only open its own sketches and other files ending in .ino or .pde.

And then the arduino IDE opens with the code.

Thank you

Best regards

Rubens.
[ArdublocklySketch.ino.tar.gz](https://github.com/carlosperate/ardublockly/files/165130/ArdublocklySketch.ino.tar.gz)
[Pythonshell_ArdublocklyServer_Ubuntu.pdf](https://github.com/carlosperate/ardublockly/files/165131/Pythonshell_ArdublocklyServer_Ubuntu.pdf)

carlosperate, 2016-03-09:
I see, you are using Arduino 1.0.5, which does not support the command line interface. You will need 1.6 or higher (I would recommend the latest one available from the Arduino website).

RubensQRZ, 2016-03-09:
thank you so much,

carlosperate, 2016-03-09:
No problem, glad you got it working.

closed 2016-03-09 by carlosperate