carlosperate/ardublockly #55: Merge blockly master

bmcage, 2016-03-07:
I track blockly master. I see some bugs were fixed that are present in ardublockly.
Due to the static typing stuff, I can't make out though what is a change in blockly master, and what is an update/bug fix in blockly master. Am I correct in assuming some changes too in zoom possibilities, trash can, and perhaps others?

Is it possible to keep on supporting the changes in blockly as present in ardublockly, would it not be good to contribute in blockly directly? I'm somewhat afraid of using ardublockly version, and not having the goodies Google might introduce in blockly.

Suggestion: 

1. can the specific ardublockly changes be indicated with comment blocks?

2. can the specific ardublockly static type values be via a flag, so that upstream contribution is possible? On glance it seems well coded and well integated, so I believe merge should be possible. Perhaps the type.js, types.js and static_typing.js in subdir core/typing ?

With above, other developers should be able to quickly determine what blockly changes can be merged in ardublockly, and what hits the new code paths present in ardublockly.

As a specific remark on static typing, if no type, void is used in Arduino. It would be better that int is used in those cases for Arduino code, less possibility for bugs. So some default type for null, and actual void in Arduino as a type 'None' or 'NoType' ?

carlosperate, 2016-03-07:
If you have a look at the commit history, you can easily track when was the last time Blockly master was merged into Ardublockly. For instance, the last catch up was done in https://github.com/carlosperate/ardublockly/commit/ec78fb902c9943e44d02f7657a2cbabc961bf62b .

I used to sync more often, when changes were relatively small. Currently the master branch is getting larger merges from the 2016 development branch and I usually spend a little bit of time reviewing the code before I pull from upstream, just to make sure things would still work as expected and no Ardublockly-specific changes are reverted somehow. In fact this weekend I did just that (no conflicts found) and was planning to do the merge tonight.

The general summary of the Ardublockly changes is listed in the blockly folder [README file](https://github.com/carlosperate/ardublockly/blob/master/blockly/README.md), but if you want to do a proper comparison I would suggest to git clone Blockly on a separe diretcory, check out the last commit  merged (in this case https://github.com/google/blockly/commit/dc1f03a168ecb5873522c3fe232938ec728f3349, as indicated in https://github.com/carlosperate/ardublockly/commit/ec78fb902c9943e44d02f7657a2cbabc961bf62b) and use a comparison tool. Is the only way to really check with complete certainty any code changes and while keeping the codebase clean.

> can the specific ardublockly static type values be via a flag, so that upstream contribution is possible? On glance it seems well coded and well integated, so I believe merge should be possible. Perhaps the type.js, types.js and static_typing.js in subdir core/typing ?

I'm not quite sure how a flag would help with merging upstream code as it will still be present in the codebase and git conflicts could still occur. Do you mean you would like to remove all the Ardublockly features and use "plain blockly" with the new blocks? Unfortunately, that would not really be possible (unless you use BlocklyDuino blocks, but then you could just update the blockly version in BlocklyDuino), as the Arduino blocks depend on these features to be present.

> Is it possible to keep on supporting the changes in blockly as present in ardublockly, would it not be good to contribute in blockly directly? I'm somewhat afraid of using ardublockly version, and not having the goodies Google might introduce in blockly.

I have submitted pull request for all the bug fixes implemented in Ardublockly, so all those are now part of Blockly upstream. However significant changes like the Static Typing are very specific to this implementation, and would probably require additional work to ensure full compatibility.
Apart from that you'd be able to see that, from inception, Ardublockly has stayed pretty close to Blockly upstream, and I plan to keep it that way. But of course, we never know what is going to happen in the future, so I've tried to keep all changes as "self contained" as possible, so hopefully it should not be an arduous task for other developers to sync more often.

> As a specific remark on static typing, if no type, void is used in Arduino. It would be better that int is used in those cases for Arduino code, less possibility for bugs. So some default type for null, and actual void in Arduino as a type 'None' or 'NoType' ?

Do you mean the `Blockly.Types.NULL` type? I'm not quite sure I've followed the question correctly, could you elaborate a little bit more?

bmcage, 2016-03-07:
First, I'm not a javascript coder, and only worked on this a couple of days, so I'm not into the code. Just trying to understand how things fit together.

To elaborate, I did not integrate the static typing changes in core blockly blocks for my code yet. All code seems to still works like that, except I need to hack static typing to assign a value of NUMBER
https://github.com/bmcage/blockly/blob/arduino/core/ardublockly/static_typing.js#L66
Unknown stuff then obtain an int value, which is fine in Arduino as long as you don't use it (http://stackoverflow.com/questions/4260048/c-function-defined-as-int-but-having-no-return-statement-in-the-body-still-compi)

There are errors in the branch with that, specifically created variables not showing up in variable drop down boxes. I'll think it over some, and I will probably merge your blockly changes. Nevertheless, having the code work, seems to indicate changes need not be as big to integrate static typing changes back into blockly proper.

This links back to the `Blockly.Types.NULL` question. As I have blocks that don't use static typing mixed with static typing blocks, those blocks get assigned void as type. Just like blocklyduino, always using int in that case seems a better default than void which will not even compile with Arduino. Once I move to your version of blockly it should not be an issue for me though.

I understand merging static typing upstream would be a difficult task, but refractoring the current type check system (https://developers.google.com/blockly/custom-blocks/type-checks), so that the static typing patch needed ardublockly becomes smaller, seems doable.

carlosperate, 2016-03-08:
I see what you meant now with undefined types in [bmcage/blockly/core/ardublockly/static_typing.js#L66](https://github.com/bmcage/blockly/blob/arduino/core/ardublockly/static_typing.js#L66), for that I would recommend to keep the `UNDEF` type where it was originally (as it indicates the variables has not had its type defined yet), an instead change how it gets converted into Arduino code [carlosperate/ardublockly/blockly/generators/arduino.js#L360](https://github.com/carlosperate/ardublockly/blob/master/blockly/generators/arduino.js#L360-L361). The problem with having the `NUMBER` type in the `else` statement, is that if you have a statement block that is trying to defer type assignment  to any lower block (block list starts from top to bottom), then having that `else` would just force that variable to be set to a Number.

To see what I mean you can compared in http://ardublockly.embeddedlog.com/demo/ the following two examples:

```xml
<xml xmlns="http://www.w3.org/1999/xhtml">
  <block type="variables_set" x="36" y="100">
    <field name="VAR">item</field>
    <value name="VALUE">
      <block type="variables_get">
        <field name="VAR">i</field>
      </block>
    </value>
    <next>
      <block type="variables_set">
        <field name="VAR">i</field>
        <value name="VALUE">
          <block type="variables_get">
            <field name="VAR">i</field>
          </block>
        </value>
      </block>
    </next>
  </block>
</xml>
```

vs

```xml
<xml xmlns="http://www.w3.org/1999/xhtml">
  <block type="variables_set" x="36" y="100">
    <field name="VAR">item</field>
    <value name="VALUE">
      <block type="variables_get">
        <field name="VAR">i</field>
      </block>
    </value>
    <next>
      <block type="variables_set">
        <field name="VAR">i</field>
        <value name="VALUE">
          <block type="variables_get">
            <field name="VAR">i</field>
          </block>
        </value>
        <next>
          <block type="variables_set">
            <field name="VAR">i</field>
            <value name="VALUE">
              <block type="text">
                <field name="TEXT"></field>
              </block>
            </value>
          </block>
        </next>
      </block>
    </next>
  </block>
</xml>
```

With your change, the second example would probably resolve in an `int` type for both variables.

When you say the blocks without static typing get assigned "void" as a type, do you mean the blocks get assigned `Blockly.Types.NULL`? Which would then be "translated" to `void` in the Arduino code as per https://github.com/carlosperate/ardublockly/blob/master/blockly/generators/arduino.js#L358-L359 ? That would be because the blocks don't have the `getVarType()` function and get NULL assigned https://github.com/bmcage/blockly/blob/arduino/core/ardublockly/static_typing.js#L153. That is probably the best place to change this behaviour, however for the time being I'd prefer to leave NULL as it indicates the variable has no type assigned, which would indicate a bug on the blocks or the "static typing" parsing of the variables. I do agree with you that having it fall back to `int` would be much safer, perhaps a new type just to indicate this should be created, so that a fallback can be set up on the builds, but a different C keyword is used during debugging.

About merging into upstream, I think the current static typing implementation needs a bit more testing to ensure it's stable enough. Even though I do believe Neil already has plans for this in the future, so I'm not sure what type of implementation he was thinking for Blockly proper.

bmcage, 2016-03-08:
Thanks for all the info, I'll try it out the following days.

closed 2016-03-08 by bmcage