carlosperate/ardublockly #58: Make Blink No Delay possible

bmcage, 2016-03-21:
As discussed in #54 blink no delay basic example should be possible with ardublockly.
This requires:

* assign pin to variables for easy change in the code 
* digital write with the variable
* computation with unsigned long

With additional blocks as present in my branch https://github.com/bmcage/ardublockly it can be done.  
Following xml works in Blockly4Arduino (load it up to see in: http://ingegno.be/Manuals/Blockly4Arduino/blockly4Arduino/index_en.html )

```
<xml xmlns="http://www.w3.org/1999/xhtml">
<block type="declare_var_digout" id="6%q8b#n3^N!7c`#)y}`U" x="87" y="38"><field name="NAME">ledPin</field><field name="PIN">13</field><next>
<block type="declare_var_bool" id="|3^6V_K~`#fd?Pkk)cL."><field name="NAME">ledState</field><value name="NUM">
<block type="io_highlow" id="B/oHeJd_fJ(h~~*^=~7V"><field name="STATE">LOW</field></block></value><next>
<block type="declare_var_ulong" id="yU7(UhcFbv6Z3}sn7}vL"><field name="NAME">previousMillis</field><value name="NUM">
<block type="math_number" id="wrp(0:4b{y%lERx/cL~h"><field name="NUM">0</field></block></value><next>
<block type="declare_var_ulong" id="=TvMLKGby{7H|[u[o_6;"><field name="NAME">currentMillis</field><value name="NUM">
<block type="math_number" id="6O*P-9Dn6szKHmKMjQ~3"><field name="NUM">0</field></block></value><next>
<block type="declare_var_ulong" id="%aJ(Vdy~h[Wx`UjBl0BZ"><field name="NAME">interval</field><value name="NUM">
<block type="math_number" id="-v/8VP`FahSD}6BVu6yr"><field name="NUM">1000</field></block></value><next>
<block type="variables_set" id="y1G(O}!ZE;UTc6_}d[/O"><field name="VAR">currentMillis</field><value name="VALUE">
<block type="time_millis" id="V0,v^)PicQdrj]tisWwR"></block></value><next>
<block type="controls_if" id="+Sw.3:D6t2%rz3.m5UIr"><value name="IF0">
<block type="logic_compare" id="56~(M96ko)78n]4*M.pc"><field name="OP">GT</field><value name="A">
<block type="math_arithmetic" id="@n*a0sF`o5R_M^F+|^gY"><field name="OP">MINUS</field><value name="A">
<block type="variables_get" id="m)hW;TfOqir@[5vmDD%S"><field name="VAR">currentMillis</field></block></value><value name="B">
<block type="variables_get" id=":H7J@)N|Wv#CDAG5S,Ie"><field name="VAR">previousMillis</field></block></value></block></value><value name="B">
<block type="variables_get" id="9#g:nhV+8i);wY#a_Ir*"><field name="VAR">interval</field></block></value></block></value><statement name="DO0">
<block type="variables_set" id="=f^o,g=}JP3d_ssRnGe%"><field name="VAR">previousMillis</field><value name="VALUE">
<block type="variables_get" id="{:|daM[[6rFq)%3=HSQk"><field name="VAR">currentMillis</field></block></value><next>
<block type="controls_if" id="X_)_6mR,oodqcNS9|FvK"><mutation else="1"></mutation><value name="IF0">
<block type="logic_compare" id="rqrnlV|M0lk%1*_F2L.%"><field name="OP">EQ</field><value name="A">
<block type="variables_get" id="amfcM9ujj/oyVBhj,*D."><field name="VAR">ledState</field></block></value><value name="B">
<block type="io_highlow" id="@I{N*FxG~{24;Z(Wl:F0"><field name="STATE">LOW</field></block></value></block></value><statement name="DO0">
<block type="variables_set" id="G,pNC9F/#Rt2yIp}Cgvb"><field name="VAR">ledState</field><value name="VALUE">
<block type="io_highlow" id="Zk[?dxVh7F1M~j_.szK5"><field name="STATE">HIGH</field></block></value></block></statement><statement name="ELSE">
<block type="variables_set" id="k_6[XpWhZaH2.pi2t3%["><field name="VAR">ledState</field><value name="VALUE">
<block type="io_highlow" id="3CM!gm6jsoX.~3KM@#?F"><field name="STATE">LOW</field></block></value></block></statement><next>
<block type="io_digitalwrite_var" id="q-!7Oa;Xtvhz|/[rUlnJ"><field name="PIN">ledPin</field><value name="STATE">
<block type="variables_get" id="L)p=4jGTsPZPU4f_%Ydc"><field name="VAR">ledState</field></block></value></block></next></block></next></block></statement></block></next></block></next></block></next></block></next></block></next></block></next></block></xml>
```

A likewise alternative solution needs to be added to ardublockly in some manner.

bmcage, 2016-03-21:
Save following as BlinkNoDelay.xml to upload to Blockly4Arduino branch of ardublockly.

[BlinkNoDelay.txt](https://github.com/carlosperate/ardublockly/files/182173/BlinkNoDelay.txt)

carlosperate, 2016-03-24:
This is a great example to test and showcase the changes discussed in https://github.com/carlosperate/ardublockly/pull/54, thanks for opening the issue for easy tracking.

carlosperate, 2016-03-30:
In this case I have only added `long` type instead of unsigned. There will be implicit type conversion, but 32 bit signed accuracy should be sufficient for most cases. Adding unsigned support will probably add very little benefit at this point.

bmcage, 2016-04-03:
I'm having a problem using the compressed js, addCompatibleTypes throws an error. Does not happen using uncompressed. 
Seems to be with the circular references, but I don't see the error ...

bmcage, 2016-04-03:
It seems in type.js
`compatibleTypeArray[i].constructor.name !== 'BlocklyType'`
and
` ||  compatibleType.constructor.name !== 'BlocklyType'`
do not work compressed because the name has been compressed and is no longer BlocklyType

carlosperate, 2016-04-06:
Thanks for the picking that up @bmcage , should be fixed by https://github.com/carlosperate/ardublockly/commit/04279f9a5d9a03ce5cf090b0d0e519216bcfd46c.
Sorry it's taking me a few days to responde to pending issues, I'm having some issues with my computer at home.

bmcage, 2016-04-07:
Ok, I have been testing changing my define blocks into setting variable with the cast block as you want to use as paradigm.

To assign pins in digitalwrite, I use my custom block which has a variable instead of a dropdown of pin numbers. As said in other threads, ideally in the future we have such variables as part of the dropdown. For now, I just try to see if I can do away with the defines blocks already.

There is still the issue that in the setup, lines like 
` pinMode(ledPin, OUTPUT);`
come first, so before the assignment of variable ledPin with it's value in the setup block.

In other words, if variables are used for the pin numbers, there must be a way to assign the variables their value before pinMode is called.

To test, rename file in attach to .xml and load it in http://ingegno.be/Manuals/Blockly4Arduino/blockly4Arduino/index_en.html
[BlinkNoDelay_xml.txt](https://github.com/carlosperate/ardublockly/files/208292/BlinkNoDelay_xml.txt)

Suggestion: the arduino function for assigning values in the setup part, those blocks should be output before the pinMode blocks? Is that easy to change? I don't think that can give issues elsewhere.

bmcage, 2016-04-07:
A possible 'clean' solution:
1/ a new block like the arduino block 'Arduino run first ... Arduino loop forever', but now with a define part also: block 'Arduino define up front ... Arduino run first ... Arduino loop forever'
2/In the define up front part, the same logic can then be used with Set variable to blocks, using cast for the special types.

carlosperate, 2016-04-09:
I see, that is definitely a case that I did not foresaw.

I'm not quite sure I fully understood your second proposed solution, could you elaborate a bit further?

Ideally I would prefer the solution to be completely invisible to the user, so that maintaining the current "block implementation" we could somehow work a way to intelligently place the code in the right order. I'll have to spend some time thinking about it, and as always, suggestions are more than welcome :)

bmcage, 2016-04-09:
The automatic way would be to write a pinMode where a variable is used after variable assignment. However, doing that, I assume you can have things like in setup already doing a digitalwrite, and the pinMode setting needs to have come before that then. So just saying: 'pinMode at the end', is not a solution.

My _clean solution_ is one solution which has 2 parts, not 2 different solutions :-) So a block like the setup/loop block, but now 3 parts defineUpFrong/setup/loop, in all those the same casting method is used to assign variables a value.

mgreenhough, 2020-11-20:
Any movement on this? I have to call my pins multiple times and having to look up which pin is which is a pain. Rather than just calling the name of the pin.