carlosperate/ardublockly #59: The communication with Arduino IDE of the  Ardublockly's Stand-alone version (for windows)  doesn’t work fine.

RubensQRZ, 2016-03-26:
Dear Carloks, 

I decided to create a stand-alone version ( for windows) of my fork of Ardublockly using yor wiki's “how-to”. That worked perfectly  (congratulations for the work, the result is fantastic).   But, there is a problem with the final application and I could notice that the same problem occurs in your stand-alone version of Ardublockly for Windows. When the user sends a Sketch to the arduino board, the arduino IDE  never returns an output. This way,   the application never ends de “sending  process”.   The Arduino board receives and runs the sketch, but,  the stand-alone version of Ardublockly doesn’t perceive that.This way you can´t send a new sketch to the Arduino board, in order to do that, you have to restart the application. I made the tests using machines with Windows 7 and  Windows 10 . I used the latest Arduino IDE version (1.6.8)  and version 1.5.8.

carlosperate, 2016-03-27:
Are you sure you are using the "arduino_debug.exe" file instead of the "arduino.exe"?

RubensQRZ, 2016-03-28:
Hi, Carlos, 

I was using arudino.exe, but i tried now with arduino_debug.exe and the same problem occurs.  I used a differente computer now ( with Windows 10),  and the same problem happened

Best Regards.

carlosperate, 2016-03-28:
Could you post or upload the ardublockly.log file if you are using the electron application? If not the console output would be helpful as well.

RubensQRZ, 2016-03-28:
Here it is:
[ardublockly.log.txt](https://github.com/carlosperate/ardublockly/files/192515/ardublockly.log.txt)

carlosperate, 2016-03-28:
When the load/verify/upload button is pressed it should print to the log one of the following 3 messages:
```
Uploading sketch to Arduino...
```
```
Verifying the sketch...
```
```
Opening the sketch in the Arduino IDE...
```

And I didn't see any of those in the log file, could you try again and send me the new log? hopefully that way it should have one of those messages printed, with the proceeding data retrieved directly from the Arduino IDE.

MadTooler, 2016-03-28:
Carlos, 

This seems related to the issue that I continue to have noted in issue #27.

RubensQRZ, 2016-03-29:
Hi Carlos, 

I am sending you two files.

One is the ardublockly.log file and the other is a doc where I explain you the actions that I performed in Ardublockly to produce the corresponding log.

Remembering that this problems just occur with the stand-alone version of Ardublockly (ardublockly_windows_v0.1.0.zip) . With the  developers's version (ardublockly-0.1.0.zip) everything works fine.

Best Regards

Rubens.

[ardublockly.log.txt](https://github.com/carlosperate/ardublockly/files/193889/ardublockly.log.txt)

[actions_performed.pdf](https://github.com/carlosperate/ardublockly/files/193892/actions_performed.pdf)

carlosperate, 2016-03-30:
Thanks for the detailed report Rubens, I'll try to look into it between tonight and tomorrow night.

carlosperate, 2016-04-02:
Just a quick update to let you know I've been able to replicate the issue locally, so I am getting closer to find a solution.

However, at least in my case, it seems like the Arduino IDE is compiling an uploading correctly, but it just is not able to print any contents to the console or log until the application is restarted (which might hide any issues or errors experienced as well). Are you 100% that in your case it is only able to verify? have you waited long enough to ensure it is not still compiling?

RubensQRZ, 2016-04-02:
Hi Carlos. Yes, the sketch is compiled and uploaded but nothing is printed to the console and the orange button never stops speenig, This way, the user can't send another sketch to the arduino board without restarting the application.

HsuHow, 2017-06-05:
I also encountered the same problem, the sketch can be uploaded to the Arduino board, but the IDE output has been show "waiting for IDE output...". I was using arudino_debug.exe.My system is win10 x64.

carlosperate, 2017-06-05:
Hi @HsuHow, could you try with this version?
https://s3-us-west-2.amazonaws.com/ardublockly-builds/windows/ardublockly_Windows32_2017-05-09_12.43_master_fe1a2.zip

RubensQRZ, 2017-06-05:
Hi, Carlos. I followed your steps to create the standalone aplication with this new verstion  and this time I had no problems. It worked perfectly fine. I am using windows 10 64bits and Arduino Ide 1.6.8 and 1.8.2.

carlosperate, 2017-06-05:
Thanks for the report Rubens. This new version has a different back end server, which in the older version sometimes got "stuck", there is still a little bit of work required in this area to make it a bit more robust, but I'm glad it's working better already.

HsuHow, 2017-06-06:
Yes, I tried this version, but unfortunately it appears the following error:
[image]

carlosperate, 2017-06-06:
Hi @HsuHow,
The ardublockly folder should contain a file called `ardublockly.log`. Could you delete that file, use Ardublockly up to the point where it gives you that error, and then attach the new `ardublockly.log` file here? (maybe you'll need to rename it to something like `ardublockly.log.txt` first)
That should contain a lot more information that could help me identify the problem.

HsuHow, 2017-06-06:
Hi, Carlos, is that so？
[ardublockly.log.txt](https://github.com/carlosperate/ardublockly/files/1055657/ardublockly.log.txt)

carlosperate, 2017-06-06:
Hi @HsuHow,
Yes, that would be the file. Thanks for taking the time to capture the log.
Did you try to upload the program and got the error shown in the image you linked?
[image]
From the log it look as if the application was just opened and closed.

HsuHow, 2017-06-06:
Hi Carlos，
Not at all, thanks for your reply. I pressed the verify button and got the error shown in the image.But no logs are generated. It just generated opened and closed msg.

carlosperate, 2017-06-07:
I see, I guess the server is not flushing to std out all that much, I'll try to make add some flushes and see if we can get more text coming out.
Btw, what language have you got selected in your Arduino IDE?

HsuHow, 2017-06-07:
My System language is Chinese，but My Arduino IDE language is English.

carlosperate, 2017-06-07:
In that case this might be related to https://github.com/carlosperate/ardublockly/issues/141

HsuHow, 2017-06-07:
That may be so,  but I don't know how to solve it. In a long time ago I use https://s3-us-west-2.amazonaws.com/ardublockly-builds/windows/releases/ardublockly_windows_v0.1.2.zip  is a success.

carlosperate, 2017-06-07:
Yes, sorry, that's still a pending issue that I need to investigate once I find a little bit of time.

edo-xebia, 2018-06-26:
Hi Carlos,

I see your last comment on this issue is one year old. The latest version packaged with the Windows installer still has the same problem: when you upload a program, no response is received from the Arduino IDE, and ArduBlockly gets stuck and needs to be restarted to upload a new program. Did you get into resolving this issue?

Our team evaluated ArduBlockly to see if we can use it in a one day workshop programming the Leaphy robot (https://leaphy.org/) for secondary school students, and because of this issue we decided not to use ArduBlockly but go with the Scratch based mBlock software.

If we are mistaken and you have a resolution for this issue, please let us know, so we can switch back to using and extending ArduBlockly instead.

Kind regards, Edo.