carlosperate/ardublockly #60: Problem with the responsiveness of the blocklyArea

RubensQRZ, 2016-03-26:
Dear Carlos

The new version of Ardublokckly, with the” javascript file toolbox”, have a problem with the responsiveness of the blocklyArea. The blocklyArea doesn’t respond automatically to changes in the browser’s window size. After changing the browser window aspect, the user needs to click the blocklyArea in order to  adjust it to the new window aspect.

This video shows the problem: [https://youtu.be/TrAF77-W16I]

Best Regards.

Rubens.

carlosperate, 2016-03-27:
Yeah, I've notice that too, it's been an issue for quite a while. I though it went away a while ago, but it looked it resurfaced, however I'm not quite sure if it went away at all, or I just stopped noticing...

Anyways, should be fixed by https://github.com/carlosperate/ardublockly/commit/1f194efd900f152b6d81b9974c9ad6e844c92be1 although this is something that Blockly already does in the [inject.js core file](https://github.com/carlosperate/ardublockly/blob/master/blockly/core/inject.js#L420-L421), so it does raise the question of why it's not being fired anymore and if there could be any other resize events not being handled. For now I will leave it as it is, and look into it againif any other issue appears.

Btw, when you say "javascript file toolbox" did you mean the change to move from loading the toolbox from an xml file to javascript file? If so that should not really affect the blockly itself, as the difference is how the toolbox file is read and parsed. Do let me know if you are referring to something else though.

RubensQRZ, 2016-03-28:
Hi Carlos, 

Yes, I was talking about the change to move from loading the toolbox from the xml file to the javascript file. I used that as a reference because the version of ardublockly that I was uising before ( with the xml file) did not have this repsonsivness problem. But now I understand that it is a coincidence. Thanks.

RubensQRZ, 2016-03-28:
I did the change you told me to do in the ardublockly/ardublockly_blockly.js  file and now the responsiveness is working fine,

Thanks a lot.

carlosperate, 2016-03-28:
No problem.

Out of curiosity, are you synchronising the Ardublockly updates manually? Might be easier to merge all the new code if you "github forked" (or just clone and track using just git) this repository. You can pull from upstream individual commits if you are concern with conflicts.

closed 2016-03-28 by carlosperate