carlosperate/ardublockly #63: Why ardublockly can NOT upload code to arduino board?

iszufu, 2016-04-08:
When I run the core version of ardublockly and upload arduino code to a Uno board, there is no any response in the server side. BTW, I have modified the ServerCompilerSettings.ini file to be suitable for Uno.

carlosperate, 2016-04-08:
So you are running the Python server and it does not respond when trying to load an sketch?
Could you post here the console output printed when trying to upload a sketch?
Which operating system and python version are you using?

carloslfu, 2016-05-20:
I have the same issue when trying to load an sketch, tested on windows 7 ultimate sp1 64 bits and Python 2.7.11. The Python server does not respond and the browser displays `net::ERR_EMPTY_RESPONSE`.

The server is runnig and responds corectly when the arduino is disconnected:

[image]

When I connect the arduino the code is loaded correctly and works well in the arduino board, but the Python server does not respond to the client and blocks the play button:

[image]

carlosperate, 2016-06-02:
I'll see if I can investigate this weekend, sorry it's taken me so long to get around it.