carlosperate/ardublockly #64: mixly compatibility

bmcage, 2016-04-08:
I heard from the Chinese front. They also make a blockly version, mixly. 
See: https://github.com/xbed/Mixly_Arduino
So, perhaps to see if some blocks are interesting, if some sharing is possible? 

I know some companies already are working on blocks to use their modules in mixly, like the grovy blocks...

carlosperate, 2016-04-09:
That's quite interesting, it looks like we are pretty much trying to achieve a versy similar goal. I'm not quite sure what Mixly itself is, are you saying they are module like grove? or that the grove stuff could work with mixly?

bmcage, 2016-04-09:
I did not look too deeply in it, but they allow for blocks for specific producer depending components. I asked a Chinese colleague, and it seems mixly is supported/pushed by a Chinese makerspace. 
I know also of one company that plans to create blocks for use in mixly.

bmcage, 2016-04-09:
Here resources of microduino for mixly, and tutorials: 
https://wiki.microduino.cc/index.php/Mixly%EF%BC%9AProject_Tutorials

carlosperate, 2016-04-10:
Right, I see, so they produce these modules and offer a blockly implementation as a companion visual editor.
Since the desktop implementation is quite different, it seems like the only this we could share are the blocks, and they have definitely quite a few more than Ardublockly. What type of collaboration did you have in mind?

bmcage, 2016-04-11:
Not sure their blocks can just be used, a lot yes, but many not. Not all seem nice blocks, a bit too close to the C code, the advantage of blockly is lost then in my opinion.
The ardublockly approach is better engineering wise :-)

Step one is contacting them so they are aware of ardublockly. As I mentioned in another thread, we need a way to have uniform blocks for blockly, but it is clear that several instances have their own approach, so to still change that .... . Companies also like to keep things internal to have a fast turn around, collaboration with OSS suffers from that.

I would suggest you reach out to mixly to know where they stand. I have indirect contact with microduino, I'll reach out to them.