carlosperate/ardublockly #65: Menu Item to load custom blocks

MadTooler, 2016-04-09:
Can a menu item be added for loading/unloading the custom blocks, such as the grove blocks?

Maybe as a sub item within the preferences or under a view menu?

Thanks.

carlosperate, 2016-04-10:
The Grove blocks and the functionality to add additional blocks (specifically extending things like types, additional rules, warnings, and things like that) are not yet ready, which is the reason a menu item hasn't been added yet.

If open the development tools and execute in the javascript console `Ardublockly.openExtraCategoriesSelect()` it should open a modal window allowing you load the current blocks.

MadTooler, 2016-04-10:
I understand.  I have been adding the Ardublockly.openExtraCategoriesSelect() line to the ardublockly.js file around line 34 so it opens at first run.

Quick question, are the json files for the additional blocks auto generated?  If not, where would I find information on how to format them?

carlosperate, 2016-04-11:
Thats the thing, it hasn't really been defined. In the future the file [/blocks/blocks_data.json](https://github.com/carlosperate/ardublockly/blob/master/blocks/blocks_data.json) will be autogenerated (not really to be kept in the repository, but the json itself would be autogenerated by the server on a specific URL) to combine the data from all the individual `blocks_config.json` files from each block subfolder.

MadTooler, 2016-04-11:
I understood about the future autogeneration of the  /blocks/blocks_data.json and that this is an experimental and not yet developed area of the program.

I was looking at the format  and function of the individual blocks_config.json files with a lo of novice confusion.  The format seems different than what the factory produces, and I was not sure where to find the details of what I actually needed to put in there and the format.

carlosperate, 2016-04-13:
Yeah, the factory in Ardublockly hasn't been modified yet, so it produces "vanilla" blockly blocks.

The details of how to create new blocks is meant to go into the wiki (which is also packed as static documentation with the Ardublockly downloads). At least one page with a general description of what is required to add new blocks still needs to be created, and then the "API" to extend Ardublockly should be defined here: https://github.com/carlosperate/ardublockly/wiki/Create-Blocks-~-Extending-Ardublockly But it does not really contain useful information yet, as this is likely to change very soon as I keep exploring the idea.