carlosperate/ardublockly #67: Translation of typeName required

bmcage, 2016-04-09:
I see quite some code checking on typeName, and not on typeKey.

I want to translate typeName, as that is always English in the currect method.
I see 

      typesArray.push([Blockly.Types[typeKey].typeName, typeKey]);

So from a selection in the dropdown of the cast block typeKey is returned. However, a grep search on typeName show it used a lot, eg in arduino.js, 'switch (typeBlockly.typeName) { '

So, how to translate. Will changing in types.js 
  typeName: 'Character',
into 
  typeName: Blockly.Msg.ARD_TYPE_CHARACTER,
and then the translations everywhere just work ?

carlosperate, 2016-04-10:
If you are refering to the TypeName from [blockly/core/type.js](https://github.com/carlosperate/ardublockly/blob/4f8dd66cfdf0f91aa3e56b1afbacfeebd007de35/blockly/core/type.js#L33), then it perhaps should not be translated, it is really meat to be an internal string used for "type" identification on Blockly. Then the code generator will translate that to whatever the code represnetation is like in [blockly/generators/arduino.js](https://github.com/carlosperate/ardublockly/blob/4f8dd66cfdf0f91aa3e56b1afbacfeebd007de35/blockly/generators/arduino.js#L370).

However, as you say, that same string is then used for that dropdown, so I think we should be deal with that translation in the in [blockly/core/types.js](https://github.com/carlosperate/ardublockly/blob/4f8dd66cfdf0f91aa3e56b1afbacfeebd007de35/blockly/core/types.js#L131) `Blockly.Types.getValidTypeArray` function and rename it to something a bit more representation of its new expanded functionality.

bmcage, 2016-04-11:
yes.
My suggestion would be to replace

    typesArray.push([Blockly.Types[typeKey].typeName, typeKey]);

by

    typesArray.push([Blockly.Types[typeKey].selectName, typeKey]);

and then when making a type selectName is translatable, so from Blockly.Msg.....

On the other hand, the switch statement on typeName could perhaps be changed to a switch based on an enum or classname or ... Once typeName is no longer exposed in the GUI,it seems a bit stupid to still use something like _typeName: 'Short Positive Number'_ in a switch statement.

carlosperate, 2016-04-11:
Yes, I like the idea of having a new attribute for the `Blockly.Type` object that contains the translated `Blockly.Msg...` string.
However, I think the name `typeName` is already the most appropriate for it, as `selectName` is to specific to the current use for it, if we then have to use the "name" for something else it'd be a bit less obviosus why it is called "select". So, we can change the constructor from:

```javascript
  this.typeName = args.typeName;
  this.compatibleTypes_ = args.compatibleTypes;
  this.generatedCheckList_ = [];
```

to:

```javascript
  this.typeId = args.typeName;
  this.typeName = Blockly.Msg...;
  this.compatibleTypes_ = args.compatibleTypes;
  this.generatedCheckList_ = [];
```

There is less than a handful of files using `typeName`, so it would be a small change to update everything with the new attributes.

bmcage, 2016-04-11:
I agree. If time over the noon break I'll give it a go tomorrow.

closed 2016-04-13 by bmcage