carlosperate/ardublockly #68: Web host/run full option

MadTooler, 2016-04-11:
Carlos,

Is it, or will it be possible to run from a www site, as with your demo, and have the local server finishing the connection to the IDE in the future?   I know your disclaimer on the online demo makes it pretty clear this is not possible right now.

I am working towards a custom mod that works with some of the hardware I have been using, and am looking for an easy way I can update features and broadcast them to others without having to get everyone to download every time.  If it is generally a bad idea to use this system in that way, please shut me down before I make too much of a mess thinking about it and wasting your time.  Thanks.

carlosperate, 2016-04-11:
It is not possible at the moment, but once we get some of the core block functionality that is currently discussed in the issues and PRs done, then I will be able to continue moving the current implementation to use websockets. This update would also allow an online version of Ardublockly to easily talk with a server running locally with very minor changes on the front end.

As far as Ardublockly priorities go, I've always wanted to be able to provide an easy to use offline version, as it was one of the original requirements when I was first developing this application (workshops run on laptops without internet connection, which is also why all python dependencies are actually included in the repository). However it is true that an easy way to update the application should be provided. I had a quick look at the Electron squirrel updater (https://github.com/electron/electron/blob/master/docs/api/auto-updater.md), but it seems like I would need to maintain 3 different update methods (the server requirements for windows are different than os x, and in parallel I would have to somehow provide packages for linux package managers to fetch new versions), of which I really do not know anything about. So, while I would love to have an autoupdater, it looks like it is not something that will happen any time soon, since other feature will probably be prioritised.

(P.S: Something like https://github.com/jenslind/electron-gh-releases seems like a cool addition to squirrel, but it needs os x signed applications and doesn't solve the linux dilemma. If anybody knows of an easy way to solve multiplatform auto-updates I'd love to hear about it.)

MadTooler, 2016-04-11:
I actually prefer your original goal of a simple offline version for my own use.  

However, I am trying to figure out the best way to a) help the even less tech savvy than myself get up and coding quickly and b) easily add functionality such as blocks without the user needing to check for updates.  It is a tough balance between what I actually prefer and see as a stronger developed skillset in the future vs. what I find my target users being more comfortable utilizing.  My target users are mostly younger kids and the parents or teachers that are tasked with getting the kids up and running.

Thanks again, Carlos, to you and the others helping develop your project.

carlosperate, 2016-04-13:
Unfortunately there is no real ETA for the move to websockets, but once that is done, you could host the Ardublockly front en on your website, updating it as necessary, and as long as the users had the server (either using Python, or the self executable packaged version) running on the computer browsing that page, then it should be able to communicate with it.

I've been thinking about a few ways of being able to update blocks and download extra blocks without updating the Ardublockly application (similar to the Arduino IDE library manager I guess), but in all honestly, I don't think I'll have the availability any time soon to implement it, as there are quite a few other things with a higher priority.