carlosperate/ardublockly #71: analogRead generates a wrong pinMode line and wrong number

bmcage, 2016-04-13:
Doing an analogRead should not generate a pinMode line, and it should be the number of the analoginput, not A0, which are the numbers for digital pins.

See: http://www.arduino.cc/en/Reference/AnalogRead

The current output of a line to read analog pin A0 and assign it to variable read is:

    int read;
    
    void setup() {
      pinMode(A0, INPUT);
    }
    
    void loop() {
      read = analogRead(A0);
    }

This compiles, but is not correct. It should be:

    int read;
    
    void setup() {
    }
    
    void loop() {
      read = analogRead(0);
    }

Doing `pinMode(A0, INPUT);` will prepare digital pin 14 (on UNO, so A0) for digital input, which is only needed for digitalRead function. Furthermore, `analogRead(A0);` is not correct, you cannot read on A0, only on 0,1,2,3,4,5 on an UNO. I would suggest to strip the A, towards users A0->A5 is nicer in the GUI, new people aften make the same mistake in the IDE of using A0.

An example to try: http://www.arduino.cc/en/tutorial/potentiometer

carlosperate, 2016-04-14:
As far as I understand `pinMode` input for an `analogRead` doesn't break anything (in fact it is required if the pin has been set as an output before). I needed to add something to the the "setup dictionary" `Blockly.Arduino.setups_` using `Blockly.Arduino.addSetup()`, to "reserve" setup code for that specific pin (so that using a different pin block does not set the pin to an output). I could have added an empty string, but I consciously decided to add that `pinMode` line to follow the same pattern than `digitalRead`, `digitalWrite`, and `analogWrite` follow. Do you know if there any disadvantage or ill effects by doing so?

About using `A0` instead of `0`, I must admit I've never given it much though. I just assumed the A0 is `#define' or global `const' somewhere to point to the right pin. Wouldn't those pin names be deal with by the preprocessor for all Arduino boards? I believe we must use A0 for digital read and write anyway, so it just seems like a more clear approach to always use `Ax` instead of the analogue number, so is using that incorrect? To me it has always seemed like an alternative notation that, not only works, but makes it more clear which pin you are using.

carlosperate, 2016-04-14:
Oh yeah, I forgot to sat, i'll check the source code, but I just assumed the pin Ax converts into gets then converted once again to the right ADC channel inside the analogRead function. Is that not something happening on all boards source code?

bmcage, 2016-04-14:
Hmm, I always did as in the docs, so never used pinMode or Ax notation for analogRead. Would have to test if it breaks anything, but reading doc further, it seems indeed to be okay.

The usecase of changing a pin from output digital, to an analog read, and hence needing setting pinMode INPUT, seems like something that requires changing wiring, so not an issue. Also something to put in the loop part anyway, which has no block at the moment.

A0 indeed works I found in the docs: https://www.arduino.cc/en/Tutorial/ReadAnalogVoltage 
For digitalread I believe you can pass 14 instead of A0, both should work.

I find it somewhat problematic to consider analogWrite/digitalRead/digitalWrite and analogRead related, they are not related, first 3 are digital, last one is analog. From a coding/learning point of view, I'm not convinced adding pinMode for analogRead is the correct thing.

Note some special usecase mentioned here: https://www.arduino.cc/en/Tutorial/AnalogInputPins
Doing pinMode(A0, INPUT) and combining with digitalWrite(A0, HIGH), creates a PULLUP, so digitalRead afterward works as PULLUP on. Ok, that is something if we want to expose it, should probably go in a specific block for it. 

So, to conclude, there is no bug. But best would be not to mix analogRead and digital on the same pin. And personally, I'd remove the pinMode setting :-)

closed 2016-04-14 by bmcage

carlosperate, 2016-04-14:
It's been a while since I've used it, but I assume we can `analogRead` and `digitalRead` from the same pin without further configuration, no?

```c
int read;

void setup() {
  pinMode(A0, INPUT);
}

void loop() {
  read = analogRead(A0);
  read = digitalRead(A0);
}
```

I think you are right about having a distinction between analogue and digital pin control, but if the above code is a possible use case, I think leaving `pinMode` in the `analogRead` block is the simplest way to ensure that the code above can be produced.

About enabling the internal pull ups, I agree, that is something that perhaps should be added as second digitalRead block.

bmcage, 2016-04-14:
Hmm, digitalRead in your example would be equal to doing

    read = analogRead(A0) > 512;

so possible, but not really needed, and if A0 is really analog and not digital, you better do a calibration on the voltage to determine if 512 is a really a good cutoff value to have HIGH as output.