carlosperate/ardublockly #74: Multiple toolboxes (Ardublockly fork)

RubensQRZ, 2016-05-02:
Dear Carlos, 

I am not sure if this is the correct place to put this question

I need your help once more.

I would like to show different toolboxes to the user depending on his choices. I would have a menu item with three options 

1 - Actuators and Sensors
2 - Robotic Car
3 - Robotic Arm

Depending of the user's choice I would decide which toolbox  (with it's specific blocks) would be shown to the user.

Another possibility  would be to have just one toolbox and enable or disable some blocks depending on the  menu item selected.

Is it possible?

Best Regards

Rubens.

carlosperate, 2016-05-03:
You can feed new toolbox xml trees into blockly at any time, the only constrain is that you cannot switch between having and not having categories.
Currently I keep a copy of the tree in "global" scope, and edit it as required in [ardublockly/ardublockly_blockly.js](https://github.com/carlosperate/ardublockly/blob/4431ecfb8dd00a223999715a9400f30628ac325e/ardublockly/ardublockly_blockly.js#L231-L256). In this case there is manual addition and removal of nodes into/from the tree, but if you have predefined toolboxes, you could just pick and chose which one to load at any given moment.

RubensQRZ, 2016-05-04:
Thank you so much.

carlosperate, 2016-05-06:
No problem.