carlosperate/ardublockly #77: Bloc define

jbet, 2016-05-08:
Hello,
I am teacher in technologie, in France, for pupil from 12 to 15 year's old.
In Semptember, I begin programmation with them because they will have Scratch to the exam.

What I need to begin.
Could you add a block  to define variable as : #define BP (digitalRead(2)).
Of course, if there is digitalRead(2), pin 2 must be an input, and BP become as an integer variable.

So I will need to have 4 blocks
For numeric input.
For analogique input.

For numeric output.
For analogique ouput.

Best regard.

bmcage, 2016-05-11:
Hmm, I have been discussing ways to have pins as variables. This is not yet in ardublockly.
My version has it but not as define. You put an UNO component to it, add a pin, and assign to the pin 'Digital Input BP'
Then BP can be used to read values. See http://ingegno.be/Manuals/Blockly4Arduino/blockly4Arduino/index_en.html

As you suggest, a preprocessor block for the code `digitalRead(2)` is bad practice, so not something you should learn children :-)

mgreenhough, 2020-11-20:
2020 and I still can not assign names to input/output pins?
A pretty basic request I would have thought?
Is there any movement on this?

bmcage, 2020-11-20:
@mgreenhough 
I have a fork which has blocks for this. Not changed in the default GUI though, but I rolled out another GUI that has those blocks expressed: https://blokkencode.ingegno.be/index_en.html

bmcage, 2020-11-20:
my fork can do this, see Example--> blink on https://blokkencode.ingegno.be/index_en.html 

Actually doing a define would be too advanced in my opinion, there comes a moment one should shift from GUI block tools to actual IDE. 
Note that avoiding delay is possible with the effect block, which works a bit like simpletimer library.

SebCanet, 2020-11-20:
Many forks of BlocklyDuino (like Ardublockly is) have this kind of enhancements, mine: [https://github.com/technologiescollege/Blockly-at-rduino](https://github.com/technologiescollege/Blockly-at-rduino)
Test which one you prefer on Internet.