carlosperate/ardublockly #79: error in source code block

bmcage, 2016-05-11:
If you add a servo block to ardublockly, you see in the left pane

    #include 

and not

    #include <Servo.h>

The correct code is passed to prettify, so this is a prettify issue. 
The fix I apply locally is:

    for (var i = 0; i < diff.length; i++) {
      if (diff[i].added) {
        resultStringArray.push(
            '<span class="code_highlight_new">' + diff[i].value.replace(/</g, '&lt;') + '</span>');
      } else if (!diff[i].removed) {
        resultStringArray.push(diff[i].value.replace(/</g, '&lt;') );
      }
    }

So `.replace(/</g, '&lt;')` added after `value`. Not sure that is the correct/best solution though ...

carlosperate, 2016-05-12:
Thank you for reporting this. It is related to https://github.com/carlosperate/ardublockly/issues/50 , and has been fixed in the latest version on https://github.com/carlosperate/ardublockly/commit/0b4e3e460a58cf2149e1e03e89de350dbfa769cd

closed 2016-05-12 by carlosperate

bmcage, 2016-05-12:
Ok, I tested it on your website, so was not aware it was fixed!

carlosperate, 2016-05-12:
Oh yeah, sorry about that, since I update it manually I only really do it for each "release". So I should have updated last weekend, but completely forgot about it, thanks for the reminder!