carlosperate/ardublockly #81: Runtime Error: Windows cannot find 'arduexec/ardublockly.exe'

jasonc1025, 2016-05-19:
Hi Carlos and Everyone Else,

I'm excited about ArduBlockly since it's based on Blockly's graphical and programming standards.  :)

As a first-timer, I'm now trying to make-my-own-block, yet I'm having difficulty with builds.  :|

After running 'python package\build_py2exe.py', and then running 'ardublockly_run.bat', I get the <Windows cannot find 'arduexec/ardublockly.exe'> error (see attached error screenshot: [1a-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf](https://github.com/carlosperate/ardublockly/files/273550/1a-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf).  

Looking into the '\arudexec\' folder only shows a '\server\' folder.  Nothing else appears in this folder. (see attached error screenshot: 
[1c-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf](https://github.com/carlosperate/ardublockly/files/273563/1c-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf).

Would 'missing modules' or 'binary dependencies' after doing a 'python package\build_py2exe.py' explain my issue? (also see attached error screenshot: 
[1b-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf] (https://github.com/carlosperate/ardublockly/files/273551/1b-FailureToRun-ardublockly.exe-missing-2016-0519-1500.pdf).

Any help would be greatly appreciated as we're developing an Arduino-Robotics curriculum for 3rd-8th grade, with a high interest to use 'ArduBlockly' as our preferred language (we've use ArduBlock in the past).  

Feel free to check out our website @ www.EandE.world, and midway on the front page is our T.A.C.H.NOLOGY®  Robot.  

Thanks in advance... :) Jason

carlosperate, 2016-05-21:
Hi Jason,

I'm glad you are finding Ardublockly useful.

The build_py2exe.py will create a self-contained executable version of the python server and place it under the `arduexec/server` directory, so it seems like that build has been successful. The messages displayed on your last screenshot would be expected (the second red box is expected for all py2exe builds, the first one is a bit confusing as it should have found those modules, but you can see a similar type of message from the [build server output](https://ci.appveyor.com/project/carlosperate/ardublockly/build/583#L506)).

The batch file created expects the electron application to be built as well (instructions described in https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly#second-step-electron-platform-independent), which is the step that will create the `ardublockly.exe` file. This is the "desktop application wrapper", and it is not strictly needed as you could have the python server running and a browser will display the application.

Taking in consideration what I have just mentioned above I realise that having this batch script generated in the build_py2exe.py script can be a bit confusing, so I should probably move that to the "pack_ardublockly.py" script instead.

You should be able to find a `arduexec/server/start.exe` file that will start the server running and open your default browser with Ardublockly.

Please let me know if you need more info or clarification.

carlosperate, 2016-06-02:
Hey @jasonc1025, I was wondering if you manage to get the application up and running?

jasonc1025, 2016-06-08:
Hi Carlos,  

Thanks for the kind follow-up.  Sorry for not replying earlier, since I've been busy preparing for our Arduino Robotics summer-courses, with hopes to successfully migrate from ArduBlock to ArduBlockly.  

Thanks for your previous support.  After further 'trial and error', I'm thankful to finally be able to mod a block and run a build on Windows, 'somewhat' successfully.  The 'somewhat' refers to some unexpected issues that I could use your help.

1.	For my first block-modding as a newbie, I made a simple change to an existing Servo to default at 10 degrees (vs. 90 degrees).  I only needed to make a change to one file: ‘ardublockly\ardublockly_toolbox.js’
```
'  <category id="catMotors" name="Motors">' +
'    <block type="servo_write">' +
'      <value name="SERVO_ANGLE">' +
'        <block type="math_number">' +
/// jwc 2016-0608-0900 '          <field name="NUM">90</field>' +
'          <field name="NUM">10</field>' +
'        </block>' +
'      </value>' +
'    </block>' +
```
2.	Then I proceeded to build, following the posted instructions as best as possible.  One issue that I encountered was that **only** one of the two ‘ardublockly_run.bat’ included my newly modded block. The root-dir version did not :(, while the auto-created ‘ardublockly_Windows32_2016-06-08_11.48’ did :).

3.	I could use your help to troubleshoot my build process.  Here is a summary of my build process with line-# referencing the attached, lengthy build log.  Kindly confirm that I my steps are complete and correctly sequenced.  :)

  a.	Line 36: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui> **cd .\package\electron**
  b.	Line 38: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui\package\electron> **npm install**
        ***	Line 40: **4 Warnings** that seem non-fatal, such as no-license field, option dep failed, no readme data
  c.	Line 69: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui\package\electron> **npm run release**
  d.	Line 128: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui> **python package\build_py2exe.py**
        ***	Line 1,819: **Fatal Error**: RuntimeError: EndUpdateResource: Access is denied.
  e.	Line 1,820: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui> **python package\build_py2exe.py**
        ***	Line 2,506: Run same command again for 2nd time and it succeeded :)
  f.	Line 2,511: C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui> **python package\pack_ardublockly.py**

Thanks again for your time.  

My students ranging from Elementary to High School will appreciate ArduBlockly, since it appears more maintainable and intuitive than the ArduBlock that we've been using in the past.  

Just to keep you posted, our website - [http://www.eande.world/](url) - has been updated regarding our Arduino T.A.C.H.NOLOGY® Robotics.  By the way, our bot was showcased at the Bay Area Maker Faire, and Massimo Banzi personally met us and gave us his positive affirmation.  :)

Looking forward in using ArduBlockly more often.  :)  Jason

[BuildLog-2016-0608-1200-02.pdf](https://github.com/carlosperate/ardublockly/files/305692/BuildLog-2016-0608-1200-02.pdf)

carlosperate, 2016-06-12:
Hi Jason,

Glad you got some progress. Let's try to cover all questions, let me know if I have left something unanswered:

>1. Then I proceeded to build, following the posted instructions as best as possible. One issue that I encountered was that only one of the two ‘ardublockly_run.bat’ included my newly modded block. The root-dir version did not :(, while the auto-created ‘ardublockly_Windows32_2016-06-08_11.48’ did :).

I'm not quite sure I understand your setup, are you trying this on a version clone from github and another downloaded prebuild? There should only be one `ardublockly_run.bat`.
Another thing that might be causing this issue is that the desktop application might be using the cached version of the javascript files. You can delete the folder `appdata` inside `arduexec` and that should remove the cache and refresh the front end completely.

>2. ... Kindly confirm that I my steps are complete and correctly sequenced. :)

>```
cd .\package\electron
npm install
npm run release
cd ../../
python package\build_py2exe.py
     *** Line 1,819: Fatal Error: RuntimeError: EndUpdateResource: Access is denied.
python package\build_py2exe.py
python package\pack_ardublockly.py
```

That should be working fine. If you want to have a look at what the build server is doing, all the commands are in the `appveyor.yml` files. It basically goes as follows:

```
cd blockly 
python build.py
cd ..\
python package\build_docs.py
python package\build_py2exe.py
cd package\electron
npm install
npm run release
cd ..\..\
python package\pack_ardublockly.py
```
You can bypass the build Blockly step at the beginning if you are using the uncompressed Blockly, as indicated in the documentation: https://github.com/carlosperate/ardublockly/wiki/Runninng-Uncompressed-Ardublockly .

Also you don't really need to pack ardublockly, as that script will basically create a copy of your ardublockly folder, remove unnecessary files and zip it all up in a file in a new folder named `releases` in the original Ardublockly folder. It's really only meant to be run on the server to prepare the zip files to be uploaded.
(so basically in your case the `pack_ardublockly.py` file copies `C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui` into a new folder named `C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\ardublockly_Windows32_2016-06-08_11.48`, clean it up and zip it into a file saved in ` C:\aa-Jwc\aa-Code-Jwc\ArduBlockly-Engine\3f-GitGui\releases\ardublockly_Windows32_2016-06-08_11.48.zip`)

Could you explain a little bit further the problem you are experiencing during the building?

jasonc1025, 2016-06-13:
Wow Carlos... my core issue was solved by your following solution:

> Another thing that might be causing this issue is that the desktop application might be using the cached version of the javascript files. You can delete the folder appdata inside arduexec and that should remove the cache and refresh the front end completely.

I'm happily closing this issue, as I'm ready to move on to attempt to add new blocks for the following features: 
- Infrared Tx/Rx Remote Control via library 'IRremote00.h', and
- LCD  2-Row Display via 'LiquidCrystal_I2C.h'.

My students and I already have it running via Arduino IDE and ArduBlock previously.  I'm not sure which is the appropriate way to ask for tips/examples to mod ArduBlockly for such features, so I created a new issue. Please clarify the better way to seek such help, and I'll be happy to comply.  :)

Thanks again for such prompt thoroughness.  My High School down to Elementary School students and their parents appreciate the progress of ArduBlockly.  :)

Special Regards...:)  Jason

closed 2016-06-13 by jasonc1025

carlosperate, 2016-06-15:
Glad it is working :)

> I'm not sure which is the appropriate way to ask for tips/examples to mod ArduBlockly for such features, so I created a new issue.

That's probably the best way to go about at the moment