carlosperate/ardublockly #84: Blank screen and nothing happens

MathijsG, 2016-05-31:
I today tried the full Windows package on a Windows 7 system.

All I get is this window:
[image]

The log says this:

`2016-05-31T12:40:25.553Z - info: [Ardublockly Electron] Ardublockly root dir: E:\Users\Mathijs Groothuis\Downloads\ardublockly
2016-05-31T12:40:25.554Z - info: [Ardublockly Electron] OS detected: win32
2016-05-31T12:40:25.555Z - info: [Server mgr] Command: E:\Users\Mathijs Groothuis\Downloads\ardublockly\arduexec\server\start.exe --findprojectroot --nobrowser
2016-05-31T12:40:26.035Z - error: [Ardublockly server] Traceback (most recent call last):
  File "C:\projects\ardublockly\ardublocklyserver\server.py", line 13, in <module>
ImportError: No module named 'BaseHTTPServer'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "start.py", line 20, in <module>
    import ardublocklyserver.server
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1191, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1161, in _load_backward_compatible
  File "C:\projects\ardublockly\ardublocklyserver\server.py", line 16, in <module>
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1191, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1161, in _load_backward_compatible
  File "C:\Python34\lib\http\server.py", line 660, in <module>
  File "C:\Python34\lib\http\server.py", line 866, in SimpleHTTPRequestHandler
  File "C:\Python34\lib\mimetypes.py", line 348, in init
  File "C:\Python34\lib\mimetypes.py", line 255, in read_windows_registry
TypeError: OpenKey() argument 2 must be str without null characters or None, not str

2016-05-31T12:40:26.047Z - info: [Ardublockly server] Process exited with code 255
2016-05-31T12:40:28.764Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:30.516Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:31.196Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:32.960Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:33.621Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:35.425Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:36.059Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:37.839Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:38.500Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:40.262Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:40.911Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:42.683Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:43.323Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:45.127Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:45.748Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:47.587Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:48.197Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:50.071Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:50.627Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:52.517Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:53.085Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:55.015Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:55.526Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:57.437Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:57.938Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:40:59.869Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:41:00.380Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:41:02.315Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:41:02.825Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:41:04.754Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-05-31T12:41:05.233Z - warn: [Ardublockly Electron] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
`

carlosperate, 2016-06-01:
That's a bit odd, the packed python server should have `BaseHTTPServer` included. Are you using the release 0.1.1?

MathijsG, 2016-06-01:
Will check later today when I'm at home. From what I remember I grabbed the latest windows package from: https://github.com/carlosperate/ardublockly/releases/

carlosperate, 2016-06-01:
Could you also make sure the zip file has been decompressed correctly? There should be a `arduexec\server\BaseHTTPServer.pyc` file in the decompressed `ardublockly` folder.

MathijsG, 2016-06-01:
I rechecked the file.

See for yourself:
[image]

carlosperate, 2016-06-02:
Sorry, since it seems that the embedded Python is looking for the `BaseHTTPServer` module I assumed you were using the 0.1.0 release. This is because one major differences between these two (which should have been invisible to the user)  was the move from Python 2 to Python 3.

So if you are definitely using v0.1.1 you shouldn’t have ` BaseHTTPServer`  in your `arduexec\server` folder, but a `server.pyc` file in the  `arduexec\server\library.zip\http\ ` directory (note the library.zip is a zip file, not a folder). Nevertheless, if you haven't done so already I would recommend to delete the original uncompressed folder and unzip it again, in case there was any error or missed file.

The odd part here is that v0.1.1 has Python 3, so the following import from `ardublocklyserver\server.py` is caught by the try/except and then while trying to import the http module it encounters the real issue.

On the mean time version 0.1.0 would probably work if you want to play around with Ardublockly, I'll look into this tonight once I get some free time at home.

carlosperate, 2016-06-02:
Okay, so first to answer your earlier question about why Ardublockly still had those paths, that's because Python is packaged within Ardublockly and those are the paths of the build server used to build all this. So having those directories in the stack strings is expected.

Unfortunately this looks like it might be a bug on Python 2.7/3.4: https://bugs.python.org/issue23604 http://bugs.python.org/issue22028
There is a discussion in the Python mailing list which describes the problem as malformed windows registry strings (in this specific instance having NULLs at the end): https://mail.python.org/pipermail/python-list/2015-January/683909.html
The thread list a couple of useful resources that apparently helped that specific user to solve the issue, but this required modifications to the registry data itself, which I do NOT recommend at all, unless you really know what you are doing.
http://www.swarley.me.uk/blog/2014/04/23/python-pip-and-windows-registry-corruption/
https://technet.microsoft.com/en-us/sysinternals/bb897448.aspx

MathijsG, 2016-06-03:
Version 1.0 is working without any problems, so I guess there is something messed up in the newer version. I will use 1.0 for now.

carlosperate, 2016-06-03:
If you replace the `arduexec/server` folder from v0.1.1, with the folder from v0.1.0, then you should have the latest Ardublockly changes with the server running Python 2 instead of Python 3. Just make sure you first delete the folder from 0.1.1 before coping the old one form 0.1.0, otherwise there might be left over files.

carlosperate, 2017-05-06:
Closing as newer version of Python in the build server should have fixed this issue.

closed 2017-05-06 by carlosperate