carlosperate/ardublockly #87: Configure Ardublockly to locate Arduino IDE

RimishRobolab, 2016-06-06:
How to configure Ardublockly to locate Arduino IDE. It's very urgent

carlosperate, 2016-06-06:
Hi @RimishRobolab,

Do you need any specific clarification about the configuration, or just a general need for step-by-step instructions? I could answer any questions you might have right now, but the instructions with screenshots might take a couple of days since I won't have much spare time until the weekend.

RimishRobolab, 2016-06-07:
Hello Sir,
I need detail step by step instructions with screenshots.
because of time constraints, as of now you can provide me step by step instruction without screenshots and after this weekend whenever you have time you can add screenshots also.

RimishRobolab, 2016-06-08:
Hello Sir, 
I did not get any answer from you. Can you please give me solution as soon as possible.
I have many queries to ask you in the pipeline.

aethaniel, 2016-06-21:
@RimishRobolab, filling the settings with the path where Arduino IDE is located is pretty straightforward.
What do you miss?

RimishRobolab, 2016-06-22:
Yes, I already got it but did not post here. When I click on path, the browse window came behind ardublockly. So I noticed it later becuase most of the time browse windows come infront of UI

carlosperate, 2016-06-22:
Yeah, sorry about that, haven't generated the "quick start" guide yet. However that specific issue is already documented in the FAQs: https://github.com/carlosperate/ardublockly/wiki/FAQs

closed 2016-11-03 by carlosperate

synetics, 2019-07-03:
Error while running start.py 
webbrowser.Error: could not locate runnable browser
This was solved by editing webbrowser.get() to webnrowser.open()
Now I am able to run start.py without error.
Thanks