carlosperate/ardublockly #90: Seeking Tips/Examples for Adding New Blocks for Infrared Tx/Rx -&- LCD-I2C

jasonc1025, 2016-06-13:
As noted earlier from a recently closed issue, I'm ready to move on to attempt to add new blocks for the following features: 
- Infrared Tx/Rx Remote Control via library 'IRremote00.h', and
- LCD  2-Row Display via 'LiquidCrystal_I2C.h'.

My students and I already have it running via Arduino IDE and ArduBlock previously.  I'm not sure which is the appropriate way to ask for tips/examples to mod ArduBlockly for such features, so I created a new issue. 

Please clarify the better way to seek such help, and I'll be happy to comply.  :)

carlosperate, 2016-06-15:
I've create a quick draft of the documentation create new blocks: https://github.com/carlosperate/ardublockly/wiki/Create-Blocks

Please have a look and let me know how it goes, or if you have any questions, and I will update it with more info as necessary.

jasonc1025, 2016-06-18:
Thanks Carlos for the helpful draft.  Regarding the wiki docs... what an honor to contribute towards this pioneering IDE.  :)

closed 2016-06-18 by jasonc1025