carlosperate/ardublockly #91: ardublockly blocks loading error

RimishRobolab, 2016-06-15:
After Runing start.py, some of blocks does not load and not generating any code in arduino source code
[image]

carlosperate, 2016-06-15:
If you press F12 on your browser it should open the developer tools, could you check the console tab and copy any errors that might be printed there?

Also, just in case, could you copy the output of the start.py console?

RimishRobolab, 2016-06-18:
[image]

RimishRobolab, 2016-06-18:
there was not Nano board option in ardublockly so I wanted to add Nano Option in settings.
I have done following changes to add Nano Option in board list

Step - 1 
	Go to ardublockly/classic/settings.html
	In the ‘Arduino Board’ selection form, a new <option> is required whose value is ‘nano’.

Step - 2
	Go To ardublocklyserver/compilersettings.py
	There is a __Arduino_type array which needs to include
	'Nano': 'arduino:avr:nano'
	as an option.
Step - 3
	Go To blockly/generators/arduino/boards.js

This defines the properties for the different support boards, so an entry for the Nano is required.  This is largely the same as the ‘uno’ entry, but the Nano has two additional analogue pins.  This is unverified, as I’m not an Arduino expert (to use at your own risk), but I believe all that is required is to create an entry as follows:

/** Arduino Nano Board Profile */
Blockly.Arduino.Boards.Profiles.nano = {
  name: 'Arduino Nano',
  description: 'Arduino Nano with ATmega328p board',
  compilerFlag: 'arduino:avr:nano',
  analogPins: Blockly.Arduino.Boards.generateAnalogIo(0, 7),
  digitalPins: Blockly.Arduino.Boards.generateDigitalIo(0, 13).concat(
                   Blockly.Arduino.Boards.generateAnalogIo(0, 7)),
  pwmPins: [['3', '3'], ['5', '5'], ['6', '6'], ['9', '9'], ['10', '10'],
            ['11', '11']],
  serial: [['serial', 'Serial']],
  serialPins: { Serial: [['RX', '0'], ['TX', '1']] },
  serialSpeed: [['300', '300'], ['600', '600'], ['1200', '1200'],
                ['2400', '2400'], ['4800', '4800'], ['9600', '9600'],
                ['14400', '14400'], ['19200', '19200'], ['28800', '28800'],
                ['31250', '31250'], ['38400', '38400'], ['57600', '57600'],
                ['115200', '115200']],
  spi: [['SPI', 'SPI']],
  spiPins: { SPI: [['MOSI', '11'], ['MISO', '12'], ['SCK', '13']] },
  spiClockDivide: [['2 (8MHz)', 'SPI_CLOCK_DIV2'],
                   ['4 (4MHz)', 'SPI_CLOCK_DIV4'],
                   ['8 (2MHz)', 'SPI_CLOCK_DIV8'],
                   ['16 (1MHz)', 'SPI_CLOCK_DIV16'],
                   ['32 (500KHz)', 'SPI_CLOCK_DIV32'],
                   ['64 (250KHz)', 'SPI_CLOCK_DIV64'],
                   ['128 (125KHz)', 'SPI_CLOCK_DIV128']],
  i2c: [['I2C', 'Wire']],
  i2cPins: { Wire: [['SDA', 'A4'], ['SCL', 'A5']] },
  i2cSpeed: [['100kHz', '100000L'], ['400kHz', '400000L']],
  builtinLed: [['BUILTIN_1', '13']],
  interrupt: [['interrupt0', '2'], ['interrupt1', '3']]
};

carlosperate, 2016-06-18:
I see, you should only have to modify the last two files you mentioned, as the dropdown with the options gets dynamically generated every time you open the settings.

Have a look at this commit and you'll see how the ESP8266 board support was added: https://github.com/carlosperate/ardublockly/commit/7cc6bc3fe310449b94e1160cc145876d82586e02

Let me know if you have any specific questions about this.

carlosperate, 2016-06-18:
I've added this article to the documentation with a general overview of how to add new boards:
https://github.com/carlosperate/ardublockly/wiki/Add-New-Arduino-Board

Please let me know if it is clear enough and/or needs more info.

carlosperate, 2016-06-19:
Btw, I had a quick look again at your error report, and I believe the issue you might have is that you are using the compressed version of Blockly. So you'll either have to use the uncompressed version (instructions can be found in the documentation in the github wiki), or you'll have to build Blockly.

RimishRobolab, 2016-06-21:
When I edited above two files and added some blocks to blockly. then I compile build.py file then it generated new compressed version of all files. so it should not be the problem.
Anyhow at the end I want the standalone ardublockly at that time I have to generate the compressed version of ardublockly.

RimishRobolab, 2016-06-21:
I tried to use ardublockly of an uncompressed version but it works fine before adding new board to ardublockly but when I add new Nano board it gives same error check below screenshot
[image]
Can You update an existing version ardublockly with Nano Board

carlosperate, 2016-06-21:
I'll see if I can add the nano profile tonight.
In the meantime make sure you are defining your board as:
```
Blockly.Arduino.Boards.profiles.nano
```
instead of
```
Blockly.Arduino.Boards.Profiles.nano
```

carlosperate, 2016-06-21:
From the general documentation I've found sounds like the nano should be compiled and uploaded like the Duemilanove, which would have the compiler flag `arduino:avr:diecimila`. Have you tried compiling with the `arduino:avr:nano` flag? does that work?

RimishRobolab, 2016-06-22:
I change the code from Blockly.Arduino.Boards.Profiles.nano to Blockly.Arduino.Boards.profiles.nano
Now it is not giving any error on console. All Blocks are visible.

1)but there is one problem when I upload sketch to arduino Nano. It shows output in Arduino IDE output as "Successfully Uploaded sketch" only. but after uploading sketch arduino nano board does nothing. means it is not uploaded in Arduino Nano Board. I have selected everything in settings options i.e. 
compiler location as arduino exe file location
sketch folder
arduino board as Nano
COM Port
Default IDE Location as compile and upload sketch.

when a sketch compile and uploaded successfully it should show message as
successfully uploaded sketch

sketch uses 846 bytes (2%) of program storage space. Maximum is 32,256 bytes. 
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.

but when I uploaded using Nano it shows message only as below
Successfully uploaded sketch

2) For Cross check, I tried fro Arduino Uno Board.
It has successfully uploaded sketch and shown output in arduino IDE Output and in arduino UNO board as well.
with output message.
successfully uploaded sketch

sketch uses 846 bytes (2%) of program storage space. Maximum is 32,256 bytes. 
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.

Did I miss anything to add in code for Arduino Nano board.

carlosperate, 2016-06-22:
> I have selected everything in settings options i.e.
compiler location as arduino exe file location

On windows make sure you are selecting the arduino_debug.exe file. With that change see what the full output says.

carlosperate, 2016-06-29:
@RimishRobolab are you still experiencing problems with this, or could I close the issue?

RimishRobolab, 2016-06-30:
Yes Sir,
This problem is solved. You can close the issue Now.

closed 2016-06-30 by carlosperate