carlosperate/ardublockly #93: Upload Sketch to Arduino

RimishRobolab, 2016-06-20:
In Ardublockly, there is Upload Sketch to Arduino Option. It takes too much time to upload sketch to arduino.
When I click Upload Sketch to Arduino option it loads arduino executable then uploads the sketch to it.
Every time you click upload sketch this process happens.
Is there any way to minimize this step. or can we upload sketch to arduino little bit fast.
because every time you upload sketch it takes too much time to upload

carlosperate, 2016-06-21:
That's the time it takes for the Arduino IDE to compile and upload the sketch.

There is one thing that could be speed up this process thoough, and that is to enable the Arduino IDE flag to keep temporary files on exit. However I haven't had a chance to test it as I am not quite sure how, when and if these temporary files are accessed on next compilation to speed up the process.

RimishRobolab, 2016-06-22:
I found one article for Arduino Speed Optimization.
http://www.instructables.com/id/Arduino-IDE-16x-compiler-optimisations-faster-code/?ALLSTEPS
But It has some disadvantages too. Read "A few mild words of caution" section in above url.

I tried instructions given in above url. it is only some microsecond faster.
Is there any other way to optimize Execution.
In ardublockly, It takes time to launch Arduino + upload and compile sketch to Arduino board.
Every time while compiling it shows arduino launch image. can we bypass this.
[image]

carlosperate, 2016-06-22:
Hi Rimish,

Thanks for the link, it was definitely an interesting read.

Ardublockly at the moment invokes the Arduino IDE and uses the Arduino console interface defined flags in order to be able to compile and upload code. There is the option of using the Arduino Builder directly for compilation, but it would requires A LOT more information about the individual board packages to be included within Ardublockly, which makes it significantly harder to add new boards. Apart from that, the output binary file needs to be uploaded, and to do that Ardublockly would have to interface with avrdude directly with all the same inconveniences described for the Arduino Builder. Both these disadvantages are worsened when considering 3rd party boards with completely different toolchains.

Changing the gcc optimisation level as described in the article would be possible, however I am not in favour Ardublockly permanently editing settings in the Arduino IDE that will affect its use outside of Ardublockly. Alternatively, as a user you can change this settings manually which could improve the compilation time. However, it sounds like you've tried this with minimal improvements, which is a bit disappointing.

Your other suggestion, to leave the application open, would mean a completely different implementation of how Ardublockly interfaces with the IDE, faking user input to paste the code and compile. It would be possible, but the benefits would be minimal for what would feel like a bit of a hack, specially when a nice and clean command line implementation is provided by the IDE.

The only option that I am aware of right now while using the Arduino IDE CLI, would be to manually set the build folder to a known location (so create and maintain from Ardublockly a OS-specific temporary folder) and then set configure the IDE to not delete temporary files on exit and manually try to clean up the folder on Ardublockly exit. In theory it should speed up compilation as only the new code should be compiled, but it is something I haven't had the chance to implement and test.

More info can be found in: https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

RimishRobolab, 2016-06-24:
Can You help me to do this?

carlosperate, 2016-06-26:
I could give you some guidance with the implementation yes, as a lot of the work will be testing it as well.
Give it a go and send me a PR and I can review it and guide you to the right direction.

RimishRobolab, 2016-07-06:
In 1 paragraph you said
There is the option of using the Arduino Builder directly for compilation, but it would requires A LOT more information about the individual board packages to be included within Ardublockly, which makes it significantly harder to add new boards. Apart from that, the output binary file needs to be uploaded, and to do that Ardublockly would have to interface with avrdude directly with all the same inconveniences described for the Arduino Builder.

In Last paragraph you said
The only option that I am aware of right now while using the Arduino IDE CLI, would be to manually set the build folder to a known location (so create and maintain from Ardublockly a OS-specific temporary folder) and then set configure the IDE to not delete temporary files on exit and manually try to clean up the folder on Ardublockly exit. In theory it should speed up compilation as only the new code should be compiled, but it is something I haven't had the chance to implement and test.

So We have two option right now to speed up ardublockly compilation. so which one of the above should we do or which one of this two will have best results for optimization