carlosperate/ardublockly #96: Ardublockly server and Upload sketch error

RimishRobolab, 2016-06-23:
1)When I Run Ardublockly, Initially It gives me error in browser console.
[image]
It gives no error on command prompt where I run server
[image]

2) I have designed one sensor block. When I drag it on editor and click on Upload and compile sketch
It gives me error in browser console. It launches Arduino software but does not upload code to it.
[image]
Command Prompt Error where I have run server
[image]

carlosperate, 2016-06-23:
This is an issue with Chrome that should be delivered soon: https://bugs.chromium.org/p/chromium/issues/detail?id=593273

carlosperate, 2016-06-23:
About the second error I will need more info than that, since it is a custom block. Could you please provide the Arduino generated code at the very least? Having the block definition and generators would be useful as well.

RimishRobolab, 2016-06-23:
I tried Using arduino_debug.exe it giving me same error
Block Definition for Sensor(sensor.js)
'use strict';

goog.provide('Blockly.Blocks.sensor');

goog.require('Blockly.Blocks');
goog.require('Blockly.Types');

Blockly.Blocks.io.HUE = 20;

Blockly.Blocks['flex_sensor'] = {
  init: function() {
    this.appendDummyInput()
        .appendField("FLEX SENSOR")
        .appendField("Connector")
        .appendField(new Blockly.FieldDropdown([["1", "1"], ["2", "2"], ["3", "3"]]), "connectorselection")
        .appendField("Operater")
        .appendField(new Blockly.FieldDropdown([["<", "<"], [">", ">"], ["==", "=="]]), "operaterselection");
    this.appendDummyInput()
        .appendField("resistance")
        .appendField(new Blockly.FieldTextInput("Enter Value between 0 - 1023"), "resistance");
    this.appendStatementInput("doflex")
        .setCheck(null);
    this.setColour(20);
    this.setTooltip('');
    this.setHelpUrl('http://www.robolab.in/');
  }
};

Block Generated Code(sensor.js)

'use strict';

goog.provide('Blockly.Arduino.sensor');

goog.require('Blockly.Arduino');

Blockly.Arduino['flex_sensor'] = function(block) {
  var connector_sel = block.getFieldValue('connectorselection');
  var operator = block.getFieldValue('operaterselection');
  var resistance = block.getFieldValue('resistance');
  var statements_name = Blockly.Arduino.statementToCode(block, 'doflex');
  // Include Macrodefinitions
  var flexcode = '#define flexPin_1 A0\n' + '#define flexPin_2 A2\n' + '#define flexPin_3 A5\n';
  var sensormacro = 'flexsensor';
  Blockly.Arduino.addInclude(sensormacro, flexcode);
  // Declare Global Variable
  var flexvariable = 'int connector_sel = 0;\n' + 'char operator;\n' + 'char resistance;\n';
  var sensorvariable = 'variable';
  Blockly.Arduino.addVariable(sensorvariable, flexvariable, false);
  // Add code to Setup function
  var flexsetup = 'Serial.begin (9600);\n' + 'pinMode(flexPin_1, INPUT);\n' + 'pinMode(flexPin_2, INPUT);\n' + 'pinMode(flexPin_3, INPUT);\n';
  var flexsetuptag = 'flexsettag';
  Blockly.Arduino.addSetup(flexsetuptag, flexsetup, false);
  // Add code to Loop function
  var code = 'connector_sel = ' + connector_sel + '\n' + 'if(connector_sel == 1)\n' +
            '{\n' +
                'int flexposition = analogRead(flexPin_1);\n' +
                'delay(1);\n' +
                'if(flexposition' + ' ' + operator + ' ' + resistance + ')\n' +
                '{\n' + statements_name + '\n}\n' + '}\n' +
            'else if(connector_sel == 2)\n' +
            '{\n' + 'int flexposition = analogRead(flexPin_2);\n' +
                'delay(1);\n' +
                'if(flexposition'+ ' ' + operator + ' ' + resistance + ')\n' +
                '{\n' + statements_name + '\n}\n' + '}\n' +
            'else if(connector_sel == 3)\n' +
            '{\n' + 'int flexposition = analogRead(flexPin_3);\n' +
                'delay(1);\n' +
                'if(flexposition' + ' ' + operator + ' ' + resistance + ')\n' +
                '{\n' + statements_name + '\n}\n' + '}\n';
  return code;
};

carlosperate, 2016-06-23:
I hope you don't mind I edited your comment to format the code correctly (you can use the github comment editor to achieve the same effect)

> I tried Using arduino_debug.exe it giving me same error

Of course, this error is not related to the version of the .exe used, it's just that you will need the debug.exe version to get proper console output displayed in Ardublockly and captured by the log.

But most importantly, could you please copy and paste the Arduino generated code on the Ardublockly application that causes this issue?

RimishRobolab, 2016-06-23:
#define flexPin_1 A0
#define flexPin_2 A2
#define flexPin_3 A5

int connector_sel = 0;
char operator;
char resistance;

void setup() {
   Serial.begin (9600);
   pinMode(flexPin_1, INPUT);
   pinMode(flexPin_2, INPUT);
   pinMode(flexPin_3, INPUT);
}

void loop() {
  connector_sel = 1
  if(connector_sel == 1)
  {
      int flexposition = analogRead(flexPin_1);
      delay(1);
     if(flexposition < Enter Value between 0 - 1023)
     {
              //statement code goes here
             //you can attach led block here
      }
  }
  else if(connector_sel == 2)
  {
         int flexposition = analogRead(flexPin_2);
        delay(1);
        if(flexposition < Enter Value between 0 - 1023)
       {
                //statement code goes here
             //you can attach led block here
        }
  }
  else if(connector_sel == 3)
  {
               int flexposition = analogRead(flexPin_3);
               delay(1);
               if(flexposition < Enter Value between 0 - 1023)
              {
                     //statement code goes here
             //you can attach led block here
              }
    }
}

carlosperate, 2016-06-23:
Nothing quite obvious there, I'm going to need to debug this locally. Do you have a full source code copy I could just copy and test in my computer? It doesn't look like you are using github for your version control system.

RimishRobolab, 2016-06-23:
You can download the source code on this link https://drive.google.com/open?id=0B9YgAk1hTD6gdGw3ZDl6YWlNd3c

carlosperate, 2016-06-23:
does this happen every time? I tried to replicate using the blocks from below and it works fine on 

`Running Python 2.7.10 (32 bit) on Windows-7-6.1.7601-SP1` (64-bit windows)

```xml
<xml xmlns="http://www.w3.org/1999/xhtml">
  <block type="flex_sensor" id="xR|D4kkkWG1M]f;MY*Eb" x="55" y="90">
    <field name="connectorselection">1</field>
    <field name="operaterselection">&lt;</field>
    <field name="resistance">Enter Value between 0 - 1023</field>
    <statement name="doflex">
      <block type="io_digitalwrite" id="e1;[XGTa/5q6DkWhZ/CD">
        <field name="PIN">13</field>
        <value name="STATE">
          <block type="io_highlow" id="?jo-DvY_*8kS}ny^(9,0">
            <field name="STATE">HIGH</field>
          </block>
        </value>
      </block>
    </statement>
  </block>
</xml>
```

carlosperate, 2016-06-23:
Hold, which language have you set up your Arduino IDE?

RimishRobolab, 2016-06-24:
I am running Python 2.7.11 (32 bit) on 64 bit windows 10 OS.
I am using an english language In Arduino IDE.
I am using Arduino Nano Board which was not included in ardublockly then I write the code for adding Nano board
If I upload only led block then it works fine but when upload my custom made sensor block then it gives me error
[image]
[image]

carlosperate, 2016-06-24:
The issue is not with the front end javascript code, but with the server parsing of the sketch file, so the console output is the part I need to be able to debug this.
I've been using your zipped version of Ardublockly and having a hard time replicating this, the application seems to work on Win7 64bits Py2.7.10 and Win10 64 bits Py2.7.11 both using Arduino 1.6.9.

Does the python output always give you the same error?

Which version of the Arduino IDE are you using? Does the same problem happen if using `arduino.exe` and `arduino_debug.exe`?

When you the error happens, could you copy the file `ArdublocklySketch\ArdublocklySketch.ino` and upload it somewhere so that I could try that one directly?

RimishRobolab, 2016-06-24:
Hello Sir,
Code Uploaded Successfully.
thanks for help.
There was syntax error in sketch.

carlosperate, 2016-06-24:
Hi Rimish,

Yes, the generated code had a few issues, but the server was unable to process the output from the Arduino IDE to be displayed in the Ardublockly interface (it failed to parse the std error output), so this problem can still occur if there is any other issue on the compilation.

Could you try to compile the problematic sketch on the Arduino IDE without ardublockly, and then copy and paste here the Arduino IDE console ouput from the bottom of the Arduino window? Hopefully that way I could see the type of text that was causing the issue on Ardublockly.

RimishRobolab, 2016-06-27:
Actually the generated code from blocks in right panel has some syntax error. When I copy this code into Arduino Software it was giving me syntax error.
So I corrected the generated code from blocks and then uploaded again.
It has uploaded the code successfully to arduino Nano and output was getting correctly.

carlosperate, 2016-06-27:
Yes, but even if the code had a syntax error, Ardublockly should have processed the Arduino IDE output properly and displayed such error in the "Arduino IDE output" box, just like it does when the code compiles or is uploaded. Therefore, there is an issue with Ardublockly that makes it crash when this particular Arduino IDE output is processed, which is why it'd be useful if you'd be able to compile that code in your particular computer (as it seems to work in mine) and see what the Arduino Output is.

RimishRobolab, 2016-06-27:
Ok Now I got it.
So What should I do for this thing because when there is a syntax error it does not show me error in Ardublockly Output Box but when it is correct it compiles and shows output.

carlosperate, 2016-06-27:
Thanks I really appreciate your help here.
If you still have the broken implementation (I believe the zip file you linked before should be a version of the code with the issue), you could then put together the blocks causing the problem, but instead of "uploading the code" or "verifying the code", you can use the option to "Open in IDE" (the yellow button). Then there are four things that if you could capture I would find quite useful:

1) If you could "save as" the Arduino sketched opened in the Arduino IDE and save that
2) If you could try to upload the sketch and copy-paste the Arduino output from the Arduino IDE it into a text file (it should be all printed in the console box at the bottom of the Arduino IDE)
3) If you could verify the code and also copy and paste the output from the Arduino IDE as well
4) If you could find the arduino "preferences.txt" file (this link shows where you can find it: https://www.arduino.cc/en/Hacking/Preferences) and save that too

Hopefully with that information I could see what Ardublockly was processing and hopefully try to fix it, or at least be able to replicate the same problem to that I can try to debug it.

RimishRobolab, 2016-07-06:
Step 1) created sketch in ardublockly and opened it in Arduino software IDE
[image]

Step 2) Save as arduino sketch in arduino IDE.

Step 3) Upload the sketch and copied the output from console to txt file.
[console output.txt](https://github.com/carlosperate/ardublockly/files/349410/console.output.txt)

Step 4) Verify sketch and copied output from console to txt file
[verify console output.txt](https://github.com/carlosperate/ardublockly/files/349422/verify.console.output.txt)

step 5) preferences.txt file
[preferences.txt](https://github.com/carlosperate/ardublockly/files/349430/preferences.txt)

RimishRobolab, 2016-07-09:
Hello Sir, Have you looked into my above comment. I have uploaded the files you wanted

carlosperate, 2016-07-09:
Hi Rimish,
I'm currently on holiday, so unfortunately I won't be able to look into
much for a little while. I'll try to spend some time on Ardublockly as soon
as I'm back.

Regards,
Carlos

RimishRobolab, 2016-07-30:
Hi Carlos Sir,
Have you tried above uploaded files and get any solution

carlosperate, 2016-08-01:
Apologies, I must admit I had forgotten about this, I'll try to give it a good look this week.

RimishRobolab, 2016-08-01:
Hello Sir,
I have solved this issue yesterday. There was a problem with my arduino.exe file. I have downloaded the arduino software again and it works for me

closed 2016-08-06 by carlosperate