carlosperate/ardublockly #100: Invalid Colour Error

RimishRobolab, 2016-06-29:
[image]
[image]

carlosperate, 2016-06-29:
It's really hard to debug without having your code changes to Ardublockly. Based on the screenshots it sounds like you've added new blocks to the toolbox with an incorrect colour value, or that you've modified any of the motor blocks `setColour` input argument.

RimishRobolab, 2016-06-30:
Hi Sir,
Thanks for help
Problem is solved. There was a problem with servo.js file. I think by mistake I would have edited the setColour argument.

carlosperate, 2016-06-30:
Glad it is working now :)

closed 2016-06-30 by carlosperate