carlosperate/ardublockly #101: add options for pin3 and pin4 when using stepper library

Ben-at-MIAD, 2016-06-30:
I'm in the process of designing a introductory physical computing course here at the Milwaukee Institute of Art and Design. I'd really love to use Ardublockly to help get past the syntax hurdle of learning programming (most of the student will have no previous experience writing code).

I've noticed Ardublockly's implementation of the stepper library doesn't have the option of setting pin3 and pin4 for stepper function.  Possible to add this to Ardublockly?

carlosperate, 2016-07-04:
Yes, that's definitely possible. At the moment I'm in the middle of working on a couple of additional features, but perhaps I can implement it towards the end of next week of end of the following week.

If you feel comfortable writing code there is always the option to give it a go and submit a pull request, I am more than happy to give you guidance if you want to try it :)

Ben-at-MIAD, 2016-07-05:
I'm happy to give it a shot.  Point me in the right direction, and I'll see what I come up with.   I've done minimal JS work, but I know my way around Python, JSON, and CSS pretty well.

carlosperate, 2016-07-05:
I just added a new update that changes a few things around the stepper, but the addition of these fields and the generator should be compatible nevertheless.

The files you'll need to modify are the:
* [Block definition](https://github.com/carlosperate/ardublockly/blob/master/blockly/blocks/arduino/stepper.js)
* [Code generator](https://github.com/carlosperate/ardublockly/blob/master/blockly/generators/arduino/stepper.js)

There are two dropdown fields for PIN1 and PIN2, so we'll need to add another two. Ideally we should have a way to select if we want to do 2 pins or 4, so that would require a field or tickbox to mutate the block. Since I assume this is your first time doing some work with Blockly this probably sounds a bit difficult, but we can divide it in 2 or 3 incremental steps where I could guide you a bit of guidance to ensure you are going in the right direction :)

Have you worked on github or git before? If you fork this repository, and then create a new branch, as soon as you commit any changes you can then submit a PR (Pull Request) to this repository. I can then use that to review the code, or guide you if you have some specific questions. Any subsequent commits to the same branch then appear in the same PR automatically and we can do this incremental work if you want to.

The first step would be to add two additional fields for PIN3 and PIN4 and the generators to create the appropriate code. This [wiki page](https://github.com/carlosperate/ardublockly/wiki/Create-Blocks) is basically a draft, but might contain useful basic background info and links to the official Blockly documentation.

carlosperate, 2017-05-06:
The stepper block has been extended to support 4 pins in https://github.com/carlosperate/ardublockly/pull/157.

closed 2017-05-06 by carlosperate