carlosperate/ardublockly #102: Ardublockly Building

RimishRobolab, 2016-07-03:
I want to create ardublockly exe file. I tried documentation from below link https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly.
I have installed everything and then did 1st step python server (platform dependent)
It has created batch file in root directory and arduexec folder.
but inside arduexec folder, there is only a server folder. there is not any ardublockly.exe present.

carlosperate, 2016-07-03:
You need to continue with the instructions to get the exe file created on the right location, you will need node with npm for that.

closed 2016-07-03 by carlosperate

RimishRobolab, 2016-07-04:
I already Installed python, pip, py2exe, Mkdocs, node.js
step 1) for windows build
python package\build_py2exe.py
It give me following error
[image]

[image]