carlosperate/ardublockly #103: Ardublockly Building Error

RimishRobolab, 2016-07-04:
I already Installed python, pip, py2exe, Mkdocs, node.js
step 1) for windows build
python package\build_py2exe.py
It give me following error
[image]

[image]

carlosperate, 2016-07-04:
The first is not an error, it just checks for the build folder from a previous build (which should be cleaned out, but if something fails can stay behind) and deletes it before building again.

About the second screenshot nothing is all that out of the ordinary, I am not quite sure why it is searching for the carbon module, but almost everything else seems to be as expected. You can compare it with the output from the build server: https://ci.appveyor.com/project/carlosperate/ardublockly/build/623#L528

But once again, that has nothing to do with creating  the ardublockly.exe file, it will only create the server folder with the serve.exe file. As I explained in the previous post you need to follow the instructions with the node side and do:

```
npm install
npm run release
```

RimishRobolab, 2016-07-06:
An Executable file created successfully.
Thank you, sir.
It was so nice of you for posting an instant reply.
Should I close the issue now.

closed 2016-07-08 by carlosperate