carlosperate/ardublockly #105: building ardublockly creates 64 bit exe file does not run 32 bit windows os

RimishRobolab, 2016-07-06:
I did all the step for creating executable file. It has successfully created exe file.
when I run it on 64 bit windows os. It runs successfully. but when I run it on 32 bit windows os it gives me error.
[image]
It means whatever steps I followed from building ardublockly wiki has created ardublockly exe for 64 bit  windows os.
so how to create ardublockly exe for 32 bit windows os

carlosperate, 2016-07-08:
You will need to follow the same steps in a 32 bit version of Windows for it to create a 32bit executable. If you don't have access to all the tools in a 32 bit computer, you can always use the IE vm images from microsoft: https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

RimishRobolab, 2016-07-28:
[image]
I tried the same steps in 32 bit version of windows but result is same

carlosperate, 2016-07-29:
Are you completely sure you are running a 32 bit version of windows ? If that's the case make sure you are building from a clean project (as far as node is concerned), if you've moved your project folder from your original dev machine, it would have moved all the npm packages from the 64 bit OS.

carlosperate, 2016-08-14:
Hi @RimishRobolab,

Were you able to solve this issue?

RimishRobolab, 2016-08-15:
Yes, This issue has been solved

closed 2016-08-15 by carlosperate