carlosperate/ardublockly #106: serial monitor in ardublockly

RimishRobolab, 2016-07-06:
how to open serial monitor of arduino IDE in ardublockly

carlosperate, 2016-07-08:
You are going to be a little bit more specific. You can select the Arduino COM port in the settings for loading the program, but if you want to get the serial data you will have to use the Arduino IDE directly.

carlosperate, 2016-08-14:
I hope your question was answered, if it hasn't please feel free to reopen this issue.

closed 2016-08-14 by carlosperate

Lapjeskater, 2016-10-20:
Hey Carlosperate,
I'm using the windows (desktop) version of ardublockly to teach a group of children the basics of programming and robotics.
It would love to have a direct option to start a serial monitor from the ardublockly window, so the users don't have to do anything with the arduino program at all.

carlosperate, 2016-10-21:
Hi @Lapjeskater, I assume you mean something as described in https://github.com/carlosperate/ardublockly/issues/29 rather opening the Arduino IDE Serial monitor directly?

Lapjeskater, 2016-10-21:
Ah I didn't see that one, this issue popped up higher in the search. Any time schedule about when its going to be implemented? 
Love the program so far!