carlosperate/ardublockly #107: A Process for Community-Created Blocks to be Considered for "Official" Build Adoption?

jasonc1025, 2016-07-12:
Hi Carlos and the Rest of the ArduBlockly Community,

After many late nights, along with Carlos' amazing help and the nice ArduBlockly/Blockly infrastructure, I've been able to create a block to read an InfraredRx module, useful for basic remote-control of a simple bot. Since the received Keycodes are Hex-Based, this new block has the capability to return either a 'Long' or "String' type output. Also built-in is a debug feature using the Serial Monitor. A photo of it is shown below:
[image]

I have a big-picture question, regarding this new block or any other new block that might be useful for the community (so they don't have to go through 're-creating the wheel/block' ;), 

Is there a 'review' process to follow -- so that any created blocks from the community can be considered for a safe-inclusion into the official build? 

Thanks again for your precious time Carlos.  :)

Jason

jasonc1025, 2016-07-14:
Hi Carlos and Others,

I also need to apologize if the block's appearance looks 'unconventional'.  As a Tech Teacher for Elementary to High School, the above block is an 'Educational Version' -- embedded with key teaching points.  

Thus, I am open to any inputs on a more standardized appearance.  Perhaps, the 'read SERVO from Pin __' could be a possible template.  

For example, to handle the 8 Hex-Digit Output as either Text (String) or Large Number (Long):
- 'read INFRARED RX from Pin __ as Type: [Text|Large Number]' or 
- 'read IrRx from Pin __ as Type: [Text|Large Number]'.  

:)  Jason

jasonc1025, 2016-07-23:
Okay, here are two versions: 1) 'Educational' with Serial Monitor value printouts for debugging and 2) 'Basic'.  Here are the current screenshots...

* 1) Educational Version 
  * Note that the 'Optional Debug' checkbox has been replaced by the dropdown menu (since I could not get the checkbox to update real-time the textual-code):
[image]
[image]
[image]

* And 2) Basic Version:
[image]

Kindly provide any constructive inputs, since I wish to be compliant to any important standards. 

Thanks again Carlos for your helpful instructions for creating new blocks.  My students from Elementary to High School students (and their parents :) are very appreciative.

Look forward in understanding the process for submitting a block for adoption.  :)

carlosperate, 2016-07-26:
Hi Jason,

The plan for the blocks is two have two main sources.
1) All Arduino included library blocks (https://www.arduino.cc/en/Reference/HomePage) should go within the blockly directory, in their respective block and generator folders.
2) Any additional blocks should go into the top-level "blocks" directory.

The idea is to have only the default blocks available in the toolbox and then have a menu to select which blocks from the "blocks" directory to include as well.
Ideally, once this feature is finished, everything should be automatically scanned simply by placing the files in that folder. There is a readme in https://github.com/carlosperate/ardublockly/tree/master/blocks with bit more info.

So, in the near future, following the steps from the documentation/wiki (which still needs to be done) should allow you to easily take advantage of all the Ardublockly features, concentrating all code within those files, as shown in https://github.com/carlosperate/ardublockly/tree/master/blocks/grove

We could start adding these blocks in that directory, but because some of these features are still under work everything is a little bit "up in the air".

jasonc1025, 2016-08-11:
Thanks Carlos for the kind explanation.    I totally understand that 'things are under construction' aspect.  That's part of the 'pioneering' process.  ;)

As a positive encouragement, I've been able to test the above 'created' block with my students who are using ArduBlockly, and they (as well as their parents) are totally excited about what you've created through this Visual Programming IDE.  

I have an eighth grader who is able to tinker with the Arduino text coding, even only after a few sessions with ArduBlockly.  :)

Thanks again 'Carlos and others' for each of your hard work in programming and sharing your efforts with others like us.   :)  

Jason

closed 2016-08-11 by jasonc1025