carlosperate/ardublockly #108: New Messages for Messasges.js not working; though new messages for En.Js does work.

jasonc1025, 2016-07-12:
Hi Carlos and the Rest of the ArduBlockly Community,

After almost finishing -- as a newbie -- my first new block "Infrared Rx" (kindly see https://github.com/carlosperate/ardublockly/issues/107), I do have a follow-up technical question:

To add new messages, such as:

•	Blockly.Msg.ARD_INFRARED_READ_BLOCK_NAME = "Comms: Infrared Rx";
•	Blockly.Msg.ARD_INFRARED_READ_FROM_PIN = " * Input: Read PIN#";

I understand that I need to edit '\blockly\msg\messages.js', yet the changes do not get incorporated when running ArduBlockly, even when I do the 4-step Build according to https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly.  

I end up directly editing (hacking) the generated file '\blockly\msg\js\en.js' for the new messages to work.

Am I overlooking something?

Thanks for your time Carlos (and anyone else who may be reading this).
Regards... :) Jason

(Educational-Technology Instructor for Kids & Beyond)

carlosperate, 2016-07-18:
Hi Jason,

Sorry for the late reply, I was on holiday until recently. For the new strings to be built in all the language files and included in Ardublockly you need to build Blockly. The instruction can be found in https://github.com/carlosperate/ardublockly/wiki/Add-New-Language , as adding a new block string would follow the same steps as adding a new language. You are also right on the fact that he build instructions fail to include a step for building blockly, I will add that as soon as I get a chance.

jasonc1025, 2016-07-23:
Thanks very much again.  Your solution has been helpful.  It worked.  

Yet, every once in a while, from the following command:

`C:\Dropbox\3a-Jwc\3a-Code\ArduBlockly-Engine\3f-GitGui-CURRENT\blockly>python build.py

I experience this intermittent following build error:
>
* **Error reading .\msg\json\en.json**
* Traceback (most recent call last):
  * File "i18n\create_messages.py", line 149, in <module>
    main();   
  * File "i18n\create_messages.py", line 65, in main source_defs = read_json_file(os.path.join(os.curdir, args.source_lang_file))
  * File "C:\Dropbox\3a-Jwc\3a-Code\ArduBlockly-Engine\3f-GitGui-URRENT\blockly\i18n\common.py", line 64, in read_json_file     raise InputError(filename, str(e))
* **common.InputError: .\msg\json\en.json: Expecting property name enclosed in double quotes: line 9 column 1** (char 176) ('Error running i18n/create_messages.py: ', CalledProcessError())

But when I look into 'en.json', I cannot find the bug.  Weirdly, it deletes the body after this following header.
>	{
		"@metadata": {
			"author": "Ellen Spertus <ellen.spertus@gmail.com>",
			"lastupdated": "2016-04-14 00:12:10.784946",
			"locale": "en",
			"messagedocumentation" : "qqq"
		},

Yet after I restore the body back into 'en.json', the second 'python build.py' appears to work.  :)

Since this issue only seems intermittent and there is a work around, I'll be happy to close this issue.  Yet, I've entered the above comments in case anybody may be familiar with this unusual but tolerated behavior.  Perhaps there's something I'm missing in my process.  :)
`

closed 2016-07-23 by jasonc1025

carlosperate, 2016-07-25:
Glad it worked. I've also updated https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly to include the instructions for building Blockly.

The problem you are experiencing with the language files is a current issue with Blockly, which has been reported in https://github.com/google/blockly/issues/447

jasonc1025, 2016-07-25:
Thanks @carlosperate for the building instructions.  :)

HackInvent, 2017-03-29:
The issue sounds to be solved, but I don't see how to fix it?
When I execute the following command in windows: `c:\Python27\python.exe i18n\js_to_json.py --input_file  msg\messages.js --output_dir msg\json\`

it generates the file, but the content is the following:
`{
	"@metadata": {
		"author": "Ellen Spertus <ellen.spertus@gmail.com>",
		"lastupdated": "2017-03-29 21:05:20.611000",
		"locale": "en",
		"messagedocumentation" : "qqq"
	},

}` 
Is it normal?

HackInvent, 2017-03-29:
In windows I had a similar problem, in js_to_json.py I replaced _INPUT_DEF_PATTERN = re.compile("""Blockly.Msg.(\w*)\s*=\s*'([^']*)';?$""")  by _INPUT_DEF_PATTERN = re.compile("""Blockly.Msg.(\w*)\s*=\s*'([^']*)';?""") to match the line.

Some how it works fine in Linux, but not in windows. I think it's related to "\n" coding and how regex lib handle that in python depending on the os.