carlosperate/ardublockly #109: ardublockly error

RimishRobolab, 2016-07-28:
I have embedded blockly turtle game in ardublockly. 
I have created two one for ardublockly and one for blockly turtle game
[image]
[image]
It is running properly in blockly web version
I have created one ardublockly exe file
In ardublockly exe version it is not showing when i click blockly game tab
[image]
check code link below
https://drive.google.com/file/d/0B9YgAk1hTD6gUXB5d2Nvd3VCX00/view?usp=sharing

RimishRobolab, 2016-07-30:
I did not get the answer from anyone for this error

RimishRobolab, 2016-07-30:
Hello Carlos Sir,
Can you help me to solve this error

carlosperate, 2016-08-01:
Hi Rimish,
Do you still need help with this?

RimishRobolab, 2016-08-01:
Yes Sir.

RimishRobolab, 2016-08-02:
Hey Carlos, have you checked out this above error

RimishRobolab, 2016-08-04:
Hey Carlos, You have not given me any reply yet. Have you checked it?

carlosperate, 2016-08-05:
Finally had a chance to have a look.
You need to move the new javascript using jquery at the top of the html document to the end of the page, after the init function. This is because jquery is not loaded the same way under electron (that's why there is a ardublocly_desktop.js script).

closed 2016-08-05 by carlosperate