carlosperate/ardublockly #110: npm install step failed

RimishRobolab, 2016-07-29:
[image]

carlosperate, 2016-07-29:
The first square are not errors, just warnings. Some packages are optional, as they are OS dependent, and then other package depencies have been deprecated. Nothing we can do there.

The second is indeed an error, in this case it seems like the issue is to do with the post install script for electron-prebuild. Since version 1.3.1 has been released only a couple of days ago, the safest thing is to just try an older version for now. Are you using the 64bit version of Node? (I assume you are running Win10 64 bit), there's been an issue reported for that on the electron pre-build repository.

I just had a look at the file that sets which versions of the npm packages to install, and have updated it to be more strict. So if you update to https://github.com/carlosperate/ardublockly/commit/406e55156540304cccdb85d756799f1227c21cff hopefully it should be working again.

carlosperate, 2016-08-14:
Hi Rimish,

I was wondering if the update fixed your issue?

RimishRobolab, 2016-08-15:
This Issue has been solved

closed 2016-08-15 by carlosperate